f4exb/sdrangel #2137: Show name device preset

loewal, 2024-05-30:
Hi,

I created 3 presets: for NOAA15, 18 and 19.
However I cannot see those names in the spectrum/waterfall window.

Is it possible to show that somewhere?

Thanks.

f4exb, 2024-05-30:
There is no connection between a preset and your current configuration so there's no point... Once you loaded the preset and everything has been configured according to it you are just done with it.

loewal, 2024-05-30:
Pity! Thanks anyway.

f4exb, 2024-05-30:
You can change the title of a channel by clicking on the gearbox icon on the top left. This may help you give a more meaningful name than just "APT Demodulator" that is only a default title. If you want to see this title in the spectrum/waterfall you also have to chose "Title" in the "FScale" combo box. Maybe this is what you are really looking for?
[image]

closed 2024-07-15 by github-actions[bot]