f4exb/sdrangel #2139: dab audio stuttering on android

held-der-arbeit, 2024-05-31:
Hi,

I try to use my rtl-sdr v4 with the sdrangel android app (7.17.1-1496-ge3d862c8a from google play store), unfortunately the dab audio is stuttering. Listening to dab music is not possible. The broadcast fm demod is working and fm music can be listened to.

To exclude rtl-sdr hardware and antenna issues I downloaded the 7.21.1 sdrangel app for ubuntu 22.04. On ubuntu 22.04 dab audio is working fine. I noticed that there where some changes on the dab module in 7.20.x. Unfortunately there are no apk files for the latest releases.
When is the next release for android planned? Maybe this fixes the dab stuttering issue on android.
Is there a guide available, how to get an android apk out of the sources in case there is no release planned?

If needed I can send also an audio recording and screenshots from the android app.

Thanks in advance.

srcejon, 2024-06-03:
Some screenshots from the android app would be useful, showing your settings and signal.

What Android device are you using?

held-der-arbeit, 2024-06-03:
I'm using a Fairphone 5 with android 13, see screenshots below.
Also on a different android 14 device I have this stuttering for dab.
On both devices the AAC in the statistics only has around 60% - 70%.
I did not change the default settings from the app, only selected the channel.

[image]
[image]
[image]

I hope this helps a bit.

srcejon, 2024-06-03:
SNR looks low. Try adjusting the gain to see if that helps.

held-der-arbeit, 2024-06-03:
Gain set to max (49,6) makes no difference, also clicking AGC next to the gain bar does not improve the situation.

srcejon, 2024-06-04:
What does the signal look like on your PC that works?

held-der-arbeit, 2024-06-04:
This is how it looks on the pc:

[image]

DAB audio is perfect on the pc. Version in use is 7.21.1.

srcejon, 2024-06-05:
Not sure - one other thing to check would be if one of the CPUs is overloaded.

srcejon, 2024-06-05:
> Gain set to max (49,6) makes no difference, also clicking AGC next to the gain bar does not improve the situation.

Try turning it right down as well.

held-der-arbeit, 2024-06-05:
> > Gain set to max (49,6) makes no difference, also clicking AGC next to the gain bar does not improve the situation.
> 
> Try turning it right down as well.

So 22.9 is the point where "audio playback" starts, below that to 0 there is no audio at all.

held-der-arbeit, 2024-06-05:
> Not sure - one other thing to check would be if one of the CPUs is overloaded.

So there are 8 cores in the system and top shows the following:

`org.sdrangel %CPU 129.0
audioserver  %CPU  17.0
audioserver  %CPU  14.6`

I would not read this as overload, as there are 8 cores available.

srcejon, 2024-06-05:
Some parts of the demod are limited to a single core, unfortunately. 

Hard to say if that is the problem.

held-der-arbeit, 2024-06-06:
Hmm.
I created a flame graph using perf for the sdrangel process running on the pc, unfortunately there is no perf on my android devices. Maybe you can identify some issue on it. The process was using around 45% cpu in top.
 
[image]

How are you building the app for android? I tried to use qt creator, but unfortunately it will not compile.

Metaln00b, 2024-06-30:
I have the same problem on my Nothing Phone 2.

Bertsl, 2024-07-26:
I have the same problem on my Samsung A54 android 14  RTL SDR V4  and SDR Angel V7.21.3.1. Has anyone found a solution? Thanks

Metaln00b, 2024-07-26:
Problem is also present, if recording the Audio Output is enabled.

FKdeveloper, 2024-08-16:
I have the same problem with my OnePlus 11. It is definitely not overloaded. welle.io works without stuttering. 

Did anyone find a solution?

closed 2024-10-01 by github-actions[bot]

moolight-seashell, 2024-12-28:
same issue on my xiaomi runing lineage

orx57, 2024-12-31:
I have the same problem here.

I'm using version 7.21.3.1 on an Acer Chromebook (Intel Core i7-10510U CPU @ 1.80GHz (8 threads, 4.90 GHz) and 16 Gb RAM) running Chrome OS 131.0.6778.96 (Official Build) (64-bit).

What other information can I provide to help?