f4exb/sdrangel #2143: Unable to add a second audio input source

dforsi, 2024-06-01:
I have two sound cards (one is builtin on the motherboard of my PC and the other is embedded in my radio), I can add an audio source and select either one. I would like to add both as separate sources for differente demodulators, but the sampling device dialog allows only to add one audio source.

f4exb, 2024-06-01:
You have to right click on the "mute" button. This is generally the same with all demods with sound. Example with the SSB demod:
[image]

f4exb, 2024-06-02:
Probably due to the way the Audio plugin is handled as if it was an actual hardware device. This does not happen with KiwiSDR for example.

f4exb, 2024-06-02:
Issue is here: https://github.com/f4exb/sdrangel/blob/master/plugins/samplesource/audioinput/audioinputplugin.cpp#L96 it should be listed as a "BuiltInDevice"

commit cfc0114c (2024-06-02): Audio Input: list plugin as a built in device so that it can be instantiated multiple times. Fixes #2143
closed 2024-06-02 by f4exb