f4exb/sdrangel #2144: Segfault with --list-devices

dforsi, 2024-06-01:
Steps to reproduce:
* run: `build/sdrangel --list-devices`

This is the backtrace:
```
Thread 1 "sdrangel" received signal SIGSEGV, Segmentation fault.
0x00007ffff58db66b in QMutex::lock() () from /lib/x86_64-linux-gnu/libQt5Core.so.5
(gdb) bt
#0  0x00007ffff58db66b in QMutex::lock() () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#1  0x00007ffff5fb7e35 in  () at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#2  0x00007ffff5fb2942 in  () at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#3  0x00007ffff5fb5706 in  () at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#4  0x00007ffff5fb5b02 in QImageReader::format() const () at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#5  0x00007ffff5fd7d61 in QPixmapIconEngine::addFile(QString const&, QSize const&, QIcon::Mode, QIcon::State) () at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#6  0x00007ffff5fdaa4a in QIcon::addFile(QString const&, QSize const&, QIcon::Mode, QIcon::State) () at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#7  0x00007ffff5fdaf37 in QIcon::QIcon(QString const&) () at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#8  0x00007ffff7e18f60 in Workspace::updateStartStopButton(bool) (this=0x555555a0f490, checked=false) at /home/sdrangel/sdrgui/gui/workspace.cpp:927
#9  0x00007ffff7e19294 in Workspace::deviceStateChanged(int, DeviceAPI*) (this=0x555555a0f490, deviceAPI=0x555555fa8380) at /home/sdrangel/sdrgui/gui/workspace.cpp:959
#10 0x00007ffff7e1ed65 in QtPrivate::FunctorCall<QtPrivate::IndexesList<0, 1>, QtPrivate::List<int, DeviceAPI*>, void, void (Workspace::*)(int, DeviceAPI*)>::call(void (Workspace::*)(int, DeviceAPI*), Workspace*, void**)
    (f=(void (Workspace::*)(Workspace * const, int, DeviceAPI *)) 0x7ffff7e1919a <Workspace::deviceStateChanged(int, DeviceAPI*)>, o=0x555555a0f490, arg=0x7fffffffcfe0) at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:152
#11 0x00007ffff7e1d5f5 in QtPrivate::FunctionPointer<void (Workspace::*)(int, DeviceAPI*)>::call<QtPrivate::List<int, DeviceAPI*>, void>(void (Workspace::*)(int, DeviceAPI*), Workspace*, void**)
    (f=(void (Workspace::*)(Workspace * const, int, DeviceAPI *)) 0x7ffff7e1919a <Workspace::deviceStateChanged(int, DeviceAPI*)>, o=0x555555a0f490, arg=0x7fffffffcfe0) at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:185
#12 0x00007ffff7e1c596 in QtPrivate::QSlotObject<void (Workspace::*)(int, DeviceAPI*), QtPrivate::List<int, DeviceAPI*>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) (which=1, this_=0x555555f0b130, r=0x555555a0f490, a=0x7fffffffcfe0, ret=0x0)
    at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:418
#13 0x00007ffff5b07c32 in  () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#14 0x00007ffff6f7a26c in MainCore::deviceStateChanged(int, DeviceAPI*) (this=0x7ffff7a7a3a0 <(anonymous namespace)::Q_QGS_mainCore::innerFunction()::holder>, _t1=0, _t2=0x555555fa8380)
    at /home/sdrangel/build/sdrbase/sdrbase_autogen/EWIEGA46WW/moc_maincore.cpp:311
#15 0x00007ffff7ceeb5a in MainWindow::deviceStateChanged(DeviceAPI*) (this=0x7fffffffd5a0, deviceAPI=0x555555fa8380) at /home/sdrangel/sdrgui/mainwindow.cpp:3023
#16 0x00007ffff7cf97b7 in QtPrivate::FunctorCall<QtPrivate::IndexesList<0>, QtPrivate::List<DeviceAPI*>, void, void (MainWindow::*)(DeviceAPI*)>::call(void (MainWindow::*)(DeviceAPI*), MainWindow*, void**)
    (f=(void (MainWindow::*)(MainWindow * const, DeviceAPI *)) 0x7ffff7ceeb22 <MainWindow::deviceStateChanged(DeviceAPI*)>, o=0x7fffffffd5a0, arg=0x7fffffffd1e0) at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:152
#17 0x00007ffff7cf81ed in QtPrivate::FunctionPointer<void (MainWindow::*)(DeviceAPI*)>::call<QtPrivate::List<DeviceAPI*>, void>(void (MainWindow::*)(DeviceAPI*), MainWindow*, void**)
    (f=(void (MainWindow::*)(MainWindow * const, DeviceAPI *)) 0x7ffff7ceeb22 <MainWindow::deviceStateChanged(DeviceAPI*)>, o=0x7fffffffd5a0, arg=0x7fffffffd1e0) at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:185
#18 0x00007ffff7cf576e in QtPrivate::QSlotObject<void (MainWindow::*)(DeviceAPI*), QtPrivate::List<DeviceAPI*>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) (which=1, this_=0x555555d98a20, r=0x7fffffffd5a0, a=0x7fffffffd1e0, ret=0x0)
    at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:418
#19 0x00007ffff5b07c32 in  () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#20 0x00007ffff6f768e5 in DeviceAPI::stateChanged(DeviceAPI*) (this=0x555555fa8380, _t1=0x555555fa8380) at /home/sdrangel/build/sdrbase/sdrbase_autogen/GULY2SVRVA/moc_deviceapi.cpp:155
#21 0x00007ffff702cd05 in DeviceAPI::engineStateChanged() (this=0x555555fa8380) at /home/sdrangel/sdrbase/device/deviceapi.cpp:882
#22 0x00007ffff702f6dc in QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (DeviceAPI::*)()>::call(void (DeviceAPI::*)(), DeviceAPI*, void**)
    (f=(void (DeviceAPI::*)(DeviceAPI * const)) 0x7ffff702cce6 <DeviceAPI::engineStateChanged()>, o=0x555555fa8380, arg=0x7fffffffd360) at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:152
#23 0x00007ffff702f2c5 in QtPrivate::FunctionPointer<void (DeviceAPI::*)()>::call<QtPrivate::List<>, void>(void (DeviceAPI::*)(), DeviceAPI*, void**)
    (f=(void (DeviceAPI::*)(DeviceAPI * const)) 0x7ffff702cce6 <DeviceAPI::engineStateChanged()>, o=0x555555fa8380, arg=0x7fffffffd360) at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:185
#24 0x00007ffff702e6df in QtPrivate::QSlotObject<void (DeviceAPI::*)(), QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) (which=1, this_=0x555555ed14f0, r=0x555555fa8380, a=0x7fffffffd360, ret=0x0)
    at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:418
#25 0x00007ffff5b07882 in  () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#26 0x00007ffff6f78111 in DSPDeviceSourceEngine::stateChanged() (this=0x5555563be830) at /home/sdrangel/build/sdrbase/sdrbase_autogen/JGTBRDHS57/moc_dspdevicesourceengine.cpp:149
#27 0x00007ffff6fc5b11 in DSPDeviceSourceEngine::setState(DSPDeviceSourceEngine::State) (this=0x5555563be830, state=DSPDeviceSourceEngine::StNotStarted) at /home/sdrangel/sdrbase/dsp/dspdevicesourceengine.cpp:66
#28 0x00007ffff6fc5b8e in DSPDeviceSourceEngine::stop() (this=0x5555563be830) at /home/sdrangel/sdrbase/dsp/dspdevicesourceengine.cpp:87
#29 0x00007ffff6f4fbe9 in DSPDeviceSourceEngine::~DSPDeviceSourceEngine() (this=0x5555563be830, this=<optimized out>) at /home/sdrangel/sdrbase/dsp/dspdevicesourceengine.cpp:57
#30 0x00007ffff6fc5acc in DSPDeviceSourceEngine::~DSPDeviceSourceEngine() (this=0x5555563be830, this=<optimized out>) at /home/sdrangel/sdrbase/dsp/dspdevicesourceengine.cpp:59
#31 0x00007ffff6f4f30d in DSPEngine::~DSPEngine() (this=0x7ffff7a7a1a0 <(anonymous namespace)::Q_QGS_dspEngine::innerFunction()::holder>, this=<optimized out>) at /home/sdrangel/sdrbase/dsp/dspengine.cpp:49
#32 0x00007ffff6f4f45a in (anonymous namespace)::Q_QGS_dspEngine::innerFunction()::Holder::~Holder() [clone .lto_priv.0] () at /home/sdrangel/sdrbase/dsp/dspengine.cpp:58
#33 0x00007ffff525aa16 in __run_exit_handlers (status=0, listp=0x7ffff53f1680 <__exit_funcs>, run_list_atexit=run_list_atexit@entry=true, run_dtors=run_dtors@entry=true) at ./stdlib/exit.c:111
#34 0x00007ffff525ab4a in __GI_exit (status=<optimized out>) at ./stdlib/exit.c:141
#35 0x0000555555557526 in runQtApplication(int, char**, qtwebapp::LoggerWithFile*) (argc=2, argv=0x7fffffffdbf8, logger=0x0) at /home/sdrangel/app/main.cpp:202
#36 0x000055555555781d in main(int, char**) (argc=2, argv=0x7fffffffdbf8) at /home/sdrangel/app/main.cpp:248
(gdb) 
```

f4exb, 2024-06-02:
One should not "exit" there 
```

f4exb, 2024-06-02:
One should not exit there:
```
#35 0x0000555555557526 in runQtApplication(int, char**, qtwebapp::LoggerWithFile*) (argc=2, argv=0x7fffffffdbf8, logger=0x0) at /home/sdrangel/app/main.cpp:202
```
this causes havoc. Just return 0 and take care of the logger which will be `nullptr`.

commit 524bfd44 (2024-06-02): Do not exit after listing available devices. Fixes #2144
closed 2024-06-02 by f4exb