f4exb/sdrangel #2147: vkFFT.h no such file or directory

sirEgghead, 2024-06-02:
Compiling fresh on Ubuntu 22.10, following the compilation instructions exactly.  Using commit ade3eed.

```[ 14%] Automatic RCC for resources/webapi.qrc
[ 14%] Automatic RCC for resources/data.qrc
[ 14%] Building CXX object sdrbase/CMakeFiles/sdrbase.dir/dsp/cudavkfftengine.cpp.o
[ 14%] Building CXX object sdrbase/CMakeFiles/sdrbase.dir/dsp/vkfftengine.cpp.o
[ 14%] Building CXX object sdrbase/CMakeFiles/sdrbase.dir/dsp/vkfftutils.cpp.o
[ 14%] Building CXX object sdrbase/CMakeFiles/sdrbase.dir/audio/audiocompressor.cpp.o
[ 14%] Building CXX object sdrbase/CMakeFiles/sdrbase.dir/dsp/fftwengine.cpp.o
[ 14%] Building CXX object sdrbase/CMakeFiles/sdrbase.dir/dsp/kissengine.cpp.o
[ 14%] Building CXX object sdrbase/CMakeFiles/sdrbase.dir/audio/audiocompressorsnd.cpp.o
[ 14%] Building CXX object sdrbase/CMakeFiles/sdrbase.dir/audio/audiofifo.cpp.o
[ 14%] Building CXX object sdrbase/CMakeFiles/sdrbase.dir/audio/audiodeviceinfo.cpp.o
[ 14%] Building CXX object sdrbase/CMakeFiles/sdrbase.dir/audio/audiodevicemanager.cpp.o
[ 14%] Building CXX object sdrbase/CMakeFiles/sdrbase.dir/sdrbase_autogen/mocs_compilation.cpp.o
[ 14%] Building CXX object sdrbase/CMakeFiles/sdrbase.dir/dsp/sigmffilerecord.cpp.o
In file included from /opt/build/sdrangel/sdrbase/dsp/vkfftutils.cpp:20:
/opt/build/sdrangel/sdrbase/dsp/vkfftutils.h:24:10: fatal error: vkFFT.h: No such file or directory
   24 | #include <vkFFT.h>
      |          ^~~~~~~~~
compilation terminated.
make[2]: *** [sdrbase/CMakeFiles/sdrbase.dir/build.make:313: sdrbase/CMakeFiles/sdrbase.dir/dsp/vkfftutils.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 14%] Building CXX object sdrbase/CMakeFiles/sdrbase.dir/audio/audiofilter.cpp.o
In file included from /opt/build/sdrangel/sdrbase/dsp/cudavkfftengine.h:24,
                 from /opt/build/sdrangel/sdrbase/dsp/cudavkfftengine.cpp:20:
/opt/build/sdrangel/sdrbase/dsp/vkfftengine.h:26:10: fatal error: vkFFT.h: No such file or directory
   26 | #include <vkFFT.h>
      |          ^~~~~~~~~
compilation terminated.
In file included from /opt/build/sdrangel/sdrbase/dsp/vkfftengine.cpp:21:
/opt/build/sdrangel/sdrbase/dsp/vkfftengine.h:26:10: fatal error: vkFFT.h: No such file or directory
   26 | #include <vkFFT.h>
      |          ^~~~~~~~~
compilation terminated.
make[2]: *** [sdrbase/CMakeFiles/sdrbase.dir/build.make:285: sdrbase/CMakeFiles/sdrbase.dir/dsp/cudavkfftengine.cpp.o] Error 1
make[2]: *** [sdrbase/CMakeFiles/sdrbase.dir/build.make:299: sdrbase/CMakeFiles/sdrbase.dir/dsp/vkfftengine.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:4887: sdrbase/CMakeFiles/sdrbase.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
```

sirEgghead, 2024-06-02:
Followed the instructions at [https://vulkan.lunarg.com/doc/sdk/1.3.283.0/linux/getting_started.html](https://vulkan.lunarg.com/doc/sdk/1.3.283.0/linux/getting_started.html) through a couple hours of building and copied the lib and include files as they specified.  I don't actually see a vkFFT.h, which still leaves me with the same result.  :/

Also I tried the suggestion at #1785 and #2085 by running adding `-DVKFFT_BACKEND=0` to the cmake options.  No change, unfortunately.

srcejon, 2024-06-03:
From https://github.com/f4exb/sdrangel/pull/1779:

    cd /opt/build
    git clone https://github.com/DTolm/VkFFT.git
    cd VkFFT
    git checkout v1.3.1

Then:

    cmake -DVKFFT_INCLUDE_DIR=/opt/build/VkFFT/vkFFT/

sirEgghead, 2024-06-03:
lol apparently I missed the appropriate issue.  :)  Sounds like I went through a couple hours on the vulkan stuff for no reason.  I'll have to uninstall it.  I'm just beginning my workday.  I'll give it a go when I get home this evening and appropriately close the issue once I test.  Thanks!

sirEgghead, 2024-06-04:
Well we're making progress.  I ran into another error that I cannot dig into right at this moment.  I will try again in a while.

```[ 18%] Building CXX object sdrbase/CMakeFiles/sdrbase.dir/channel/remotedataqueue.cpp.o
/opt/build/sdrangel/sdrbase/dsp/cudavkfftengine.cpp: In member function ‘virtual void CUDAvkFFTEngine::transform()’:
/opt/build/sdrangel/sdrbase/dsp/cudavkfftengine.cpp:113:9: error: ‘PROFILER_START’ was not declared in this scope
  113 |         PROFILER_START()
      |         ^~~~~~~~~~~~~~
/opt/build/sdrangel/sdrbase/dsp/cudavkfftengine.cpp:115:9: error: ‘PROFILER_STOP’ was not declared in this scope
  115 |         PROFILER_STOP(QString("%1 TX %2").arg(getName()).arg(m_currentPlan->n))
      |         ^~~~~~~~~~~~~
/opt/build/sdrangel/sdrbase/dsp/cudavkfftengine.cpp:121:9: error: ‘PROFILER_RESTART’ was not declared in this scope
  121 |         PROFILER_RESTART()
      |         ^~~~~~~~~~~~~~~~
/opt/build/sdrangel/sdrbase/dsp/cudavkfftengine.cpp:123:82: error: ‘launchParams’ was not declared in this scope; did you mean ‘cudaLaunchParams’?
  123 |         VkFFTResult resFFT = VkFFTAppend(plan->m_app, plan->m_inverse ? 1 : -1, &launchParams);
      |                                                                                  ^~~~~~~~~~~~
      |                                                                                  cudaLaunchParams
/opt/build/sdrangel/sdrbase/dsp/cudavkfftengine.cpp:109:21: warning: unused variable ‘res’ [-Wunused-variable]
  109 |         cudaError_t res = cudaSuccess;
      |                     ^~~
/opt/build/sdrangel/sdrbase/dsp/cudavkfftengine.cpp:110:15: warning: unused variable ‘buffer’ [-Wunused-variable]
  110 |         void* buffer = ((void**)&plan->m_buffer)[0];
      |               ^~~~~~
/opt/build/sdrangel/sdrbase/dsp/cudavkfftengine.cpp:123:21: warning: unused variable ‘resFFT’ [-Wunused-variable]
  123 |         VkFFTResult resFFT = VkFFTAppend(plan->m_app, plan->m_inverse ? 1 : -1, &launchParams);
      |                     ^~~~~~
[ 18%] Building CXX object sdrbase/CMakeFiles/sdrbase.dir/channel/remotedatareadqueue.cpp.o
At global scope:
cc1plus: note: unrecognized command-line option ‘-Wno-inconsistent-missing-override’ may have been intended to silence earlier diagnostics
make[2]: *** [sdrbase/CMakeFiles/sdrbase.dir/build.make:285: sdrbase/CMakeFiles/sdrbase.dir/dsp/cudavkfftengine.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/opt/build/sdrangel/sdrbase/channel/channelapi.cpp: In member function ‘void ChannelAPI::handleInputMessages()’:
/opt/build/sdrangel/sdrbase/channel/channelapi.cpp:45:25: warning: possible problem detected in invocation of ‘operator delete’ [-Wdelete-incomplete]
   45 |                         delete message;
      |                         ^~~~~~~~~~~~~~
/opt/build/sdrangel/sdrbase/channel/channelapi.cpp:40:18: warning: ‘message’ has incomplete type
   40 |         Message* message;
      |                  ^~~~~~~
In file included from /opt/build/sdrangel/sdrbase/channel/channelapi.h:35,
                 from /opt/build/sdrangel/sdrbase/channel/channelapi.cpp:24:
/opt/build/sdrangel/sdrbase/util/messagequeue.h:29:7: note: forward declaration of ‘class Message’
   29 | class Message;
      |       ^~~~~~~
/opt/build/sdrangel/sdrbase/channel/channelapi.cpp:45:25: note: neither the destructor nor the class-specific ‘operator delete’ will be called, even if they are declared when the class is defined
   45 |                         delete message;
      |                         ^~~~~~~~~~~~~~
/opt/build/VkFFT/vkFFT/vkFFT/vkFFT_CodeGen/vkFFT_MathUtils/vkFFT_MathUtils.h: In function ‘void PfAppendConversionStart(VkFFTSpecializationConstantsLayout*, PfContainer*, PfContainer*)’:
/opt/build/VkFFT/vkFFT/vkFFT/vkFFT_CodeGen/vkFFT_MathUtils/vkFFT_MathUtils.h:165:17: warning: this statement may fall through [-Wimplicit-fallthrough=]
  165 |                 switch ((out->type % 100) / 10) {
      |                 ^~~~~~
/opt/build/VkFFT/vkFFT/vkFFT/vkFFT_CodeGen/vkFFT_MathUtils/vkFFT_MathUtils.h:194:9: note: here
  194 |         case 3:
      |         ^~~~
/opt/build/VkFFT/vkFFT/vkFFT/vkFFT_CodeGen/vkFFT_MathUtils/vkFFT_MathUtils.h: In function ‘void PfAppendConversionEnd(VkFFTSpecializationConstantsLayout*, PfContainer*, PfContainer*)’:
/opt/build/VkFFT/vkFFT/vkFFT/vkFFT_CodeGen/vkFFT_MathUtils/vkFFT_MathUtils.h:238:17: warning: this statement may fall through [-Wimplicit-fallthrough=]
  238 |                 switch ((out->type % 100) / 10) {
      |                 ^~~~~~
/opt/build/VkFFT/vkFFT/vkFFT/vkFFT_CodeGen/vkFFT_MathUtils/vkFFT_MathUtils.h:261:9: note: here
  261 |         case 3:
      |         ^~~~
At top level:
cc1plus: note: unrecognized command-line option ‘-Wno-inconsistent-missing-override’ may have been intended to silence earlier diagnostics
At top level:
cc1plus: note: unrecognized command-line option ‘-Wno-inconsistent-missing-override’ may have been intended to silence earlier diagnostics
make[1]: *** [CMakeFiles/Makefile2:5058: sdrbase/CMakeFiles/sdrbase.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
```

srcejon, 2024-06-04:
A recent tidy removed a required include - have just posted this pull request to fix it https://github.com/f4exb/sdrangel/pull/2149

sirEgghead, 2024-06-04:
You are awesome.  I'll have a go at it this evening when I get back home.

sirEgghead, 2024-06-04:
More progress.
```
[ 24%] Building CXX object sdrbase/CMakeFiles/sdrbase.dir/sdrbase_autogen/3YJK5W5UP7/qrc_webapi.cpp.o
[ 24%] Linking CUDA device code CMakeFiles/sdrbase.dir/cmake_device_link.o
nvlink warning : Skipping incompatible '/usr/lib/x86_64-linux-gnu/libdl.a' when searching for -ldl (target: sm_60)
nvlink fatal   : elfLink linker library load error (target: sm_60)
make[2]: *** [sdrbase/CMakeFiles/sdrbase.dir/build.make:3519: sdrbase/CMakeFiles/sdrbase.dir/cmake_device_link.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:5058: sdrbase/CMakeFiles/sdrbase.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
```
I'm looking into my cuda toolkit now.  It's on the latest from my distro, 12.0.140.  I might have to push it further.

closed 2024-07-21 by github-actions[bot]