f4exb/sdrangel #2150: Why does sdrangel depend on out of date libraries?

tomcass240, 2024-06-04:
I want to install sdrangel on Debian Bookworm but it isn't installable because it requires packages that aren't available on this version of Debian because they're outdated.

Is there any chance someone is going to release a version that can be installed on an operating system that isn't out of date?

srcejon, 2024-06-05:
There are flatpak and snap builds available, or you can compile from source.

f4exb, 2024-06-05:
> I want to install sdrangel on Debian Bookworm but it isn't installable because it requires packages that aren't available on this version of Debian because they're outdated.

The Docker image builds perfectly based on a Bookworm image: https://github.com/f4exb/sdrangel-docker/blob/master/sdrangel/Dockerfile

tomcass240, 2024-06-10:
> The Docker image builds perfectly based on a Bookworm image: https://github.com/f4exb/sdrangel-docker/blob/master/sdrangel/Dockerfile

How do I use the docker image?

f4exb, 2024-06-11:
The answer should be here: https://github.com/f4exb/sdrangel-docker/blob/master/sdrangel/readme.md

closed 2024-07-28 by github-actions[bot]