f4exb/sdrangel #2152: Morse Decoder not available for Windows release

krzotr, 2024-06-05:
Hi,

Recent release 7.21.1 for Windows shows:

> Morse Decoder: release for Windows
> ... **the Morse Decoder is now available in Windows.** ...

I have installed  `sdrangel-7.21.1-win64.exe` and do not see the **Morse Decoder** plugin. No **.dll** file in `SDRangel/plugins/` folder.

In [SDRangel CI/CD for Windows ](https://github.com/f4exb/sdrangel/actions/runs/9368161229/job/25789484380) log file flag `ENABLE_FEATURE_MORSEDECODER` is set to `OFF` it means plugin `Morse Decoder` is not available in 7.21.1 release.

```
2024-06-04T13:45:04.9839279Z -- Not building morsedecoder (ENABLE_FEATURE_MORSEDECODER=OFF GGMODSE_FOUND=)
```

f4exb, 2024-06-06:
yes the build still does not work

srcejon, 2024-06-06:
> yes the build still does not work... 

What's the problem? It builds for me.

f4exb, 2024-06-06:
It shows here: https://github.com/f4exb/sdrangel/actions/runs/9171799885 This is the latest job having the ENABLE_FEATURE_MORSEDECODER set to ON and the latest of the runs uses the latest copy of ggmorse with the fixes for Windows (it is my fork but still has the proper changes). 

It seems it does not put the GGMorse include directory on the compilation command line. I cannot explain this...
```
C:\PROGRA~1\MICROS~2\2022\ENTERP~1\VC\Tools\MSVC\1439~1.335\bin\Hostx64\x64\cl.exe  /nologo /TP -DAPPLICATION_NAME=\"SDRangel\" -DARCHITECTURE_x86_64 -DCOMPANY=\"f4exb\" -DCOPYRIGHT="\"Copyright (c) 2019 Edouard Griffiths. All rights reserved.\"" -DHAS_FT8 -DIDENTIFIER=\"com.f4exb.SDRangel\" -DLIB=\"lib\" -DNOMINMAX -DQT_CORE_LIB -DQT_GUI_LIB -DQT_MULTIMEDIA_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_POSITIONING_LIB -DQT_WEBSOCKETS_LIB -DQT_WIDGETS_LIB -DSDRANGEL_LIB_VERSION=\"7.20.1\" -DSDRANGEL_VERSION=\"7.20.1-31-gd30812789\" -DSDR_RX_SAMPLE_24BIT -DUSE_SSE2 -DUSE_SSE3 -DUSE_SSE4_1 -DUSE_SSE4_2 -DUSE_SSSE3 -D_CRT_SECURE_CPP_OVERLOAD_STANDARD_NAMES -D_CRT_SECURE_CPP_OVERLOAD_STANDARD_NAMES_COUNT -D_USE_MATH_DEFINES -Dfeaturemorsedecoder_EXPORTS -ID:\a\sdrangel\sdrangel\build\plugins\feature\morsedecoder -ID:\a\sdrangel\sdrangel\plugins\feature\morsedecoder -ID:\a\sdrangel\sdrangel\build\plugins\feature\morsedecoder\featuremorsedecoder_autogen\include -ID:\a\sdrangel\sdrangel\sdrbase -ID:\a\sdrangel\sdrangel\exports -ID:\a\sdrangel\sdrangel\sdrgui -ID:\a\sdrangel\sdrangel\swagger\sdrangel\code\qt5\client -ID:\a\sdrangel\sdrangel\build\external\ggmorse\src -external:IC:\Qt\5.15.2\msvc2019_64\include -external:IC:\Qt\5.15.2\msvc2019_64\include\QtCore -external:IC:\Qt\5.15.2\msvc2019_64\.\mkspecs\win32-msvc -external:IC:\Qt\5.15.2\msvc2019_64\include\QtWidgets -external:IC:\Qt\5.15.2\msvc2019_64\include\QtGui -external:IC:\Qt\5.15.2\msvc2019_64\include\QtANGLE -external:IC:\Qt\5.15.2\msvc2019_64\include\QtMultimedia -external:IC:\Qt\5.15.2\msvc2019_64\include\QtNetwork -external:IC:\Qt\5.15.2\msvc2019_64\include\QtWebSockets -external:IC:\Qt\5.15.2\msvc2019_64\include\QtPositioning -external:IC:\Qt\5.15.2\msvc2019_64\include\QtOpenGL -external:W0 /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MD /O2 /Ob2 /DNDEBUG -std:c++14 /GL /wd4996 /wd4267 /wd4305 /showIncludes /Foplugins\feature\morsedecoder\CMakeFiles\featuremorsedecoder.dir\morsedecoder.cpp.obj /Fdplugins\feature\morsedecoder\CMakeFiles\featuremorsedecoder.dir\ /FS -c D:\a\sdrangel\sdrangel\plugins\feature\morsedecoder\morsedecoder.cpp
D:\a\sdrangel\sdrangel\plugins\feature\morsedecoder\morsedecoderworker.h(23): fatal error C1083: Cannot open include file: 'ggmorse/ggmorse.h': No such file or directory
```

f4exb, 2024-06-06:
I was blind... yes it does but this is not the "include" directory:
`-ID:\a\sdrangel\sdrangel\build\external\ggmorse\src`

f4exb, 2024-06-06:
It looks like we have it. I will push a new release soon...

closed 2024-06-06 by f4exb