f4exb/sdrangel #2153: RemoteTCPInput and RTL-SDR v4 - Invalid behavior after changing gain

krzotr, 2024-06-06:
Hi,

I use `RemoteTCPInput` as a source. When connected to my `RTL-SDR v4 (RTLSDRBlog, Blog V4, SN: 00000001)` I have a problem setting gain. After changing, the label `0.0dB` is gone (maximum gain is set to 28.0 however, RTL-SDR v4 supports up to 49.6). On Nooelec v3 and Nooelec v5, everything is fine.

I have tested on different `librtlsdr`. All the same, gain settings for RTL-SDR v4 do not work correctly
- [rtl-sdr-blog](https://github.com/rtlsdrblog/rtl-sdr-blog)
- [librtlsdr](https://github.com/librtlsdr/librtlsdr)
- [librtlsdr-steve-m](https://github.com/steve-m/librtlsdr)

Nooelec v5 - Everything is fine

https://github.com/f4exb/sdrangel/assets/655681/765b6d1f-0248-4323-88fe-f23a251db996

RTL-SDR v4 - Problem with Gain

https://github.com/f4exb/sdrangel/assets/655681/189822e3-9fec-41dc-a860-40700713ea18

Sending logs: [sdrangle.log](https://github.com/user-attachments/files/15651192/sdrangle.log)

srcejon, 2024-06-14:
V4 not currently supported by RemoteTCPInput - will add support for it.

lordjalapeno, 2024-09-15:
HI there - Running SDRangel 7.220 on OrangiPI 3b / Ubuntu  and have the same issues as above with RermoteTCPInput.

srcejon, 2024-10-31:
Fixed in 7.22.1.

closed 2024-10-31 by srcejon