f4exb/sdrangel #2155: RSPdxR2 Support

alphafox02, 2024-06-07:
I'm sure it's already known, but just putting here in case, the R2 needs SDRPlay API 3.15 on Linux. SDRAngel is currently at 3.14.

f4exb, 2024-06-08:
Indeed this is mentioned here: https://www.sdrplay.com/api/

f4exb, 2024-06-08:
Dockerfile updated: https://github.com/f4exb/sdrangel-docker/commit/1dbe120e482c8676de9a84220f05dbcc7ec0cf1e

bodri82, 2024-06-08:
can you update to the regular sdrangel not docker:) so i can use rspdx-r2 i compiled whit sdrplay but dont work  tanx in advance

alphafox02, 2024-06-08:
I’m curious what the .2 change brought?

srcejon, 2024-06-08:
I've updated https://github.com/srcejon/sdrplayapi with 3.15.2 (which is used on https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux)

I've updated Mac arm64 build to use 3.15.0.

alphafox02, 2024-06-08:
Will that get sucked up into the 22.04 deb build?

bodri82, 2024-06-08:
> I've updated https://github.com/srcejon/sdrplayapi with 3.15.2 (which is used on https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux)
> 
> I've updated Mac arm64 build to use 3.15.0.

I have compiled but when I run sdrangel I can't select the antenna or change the vfo with the mouse I see the menu for the antenna but can't click

f4exb, 2024-06-08:
> Will that get sucked up into the 22.04 deb build?

It means creating a new image based on 24.04 for the Appveyor build. I suppose this is feasible but I am busy with something else at the moment. Moreover 24.04 is not yet available with `do-release-upgrade` so I suppose some fixes are still expected.

srcejon, 2024-06-08:
I've made a PR on sdrangel-windows-libraries https://github.com/f4exb/sdrangel-windows-libraries/pull/22 with the 3.15 update.

alphafox02, 2024-06-09:
Don’t suppose you could sustain dual builds and keep 22.04 going for awhile 😀

ON4KHG, 2024-06-17:
I tried the last build sdrangel-7.21.2-19-g12677b1bf-win64.exe but not possible to select the Antenna port, Gain,... of the associated RSPdx-r2. The buttons are present in SDRAngel but they are not active (nothing happens when clicking on them).
Thanks.

srcejon, 2024-06-17:
Sorry @f4exb, need one more update to sdrangel-windows-librarys: https://github.com/f4exb/sdrangel-windows-libraries/pull/23

srcejon, 2024-06-17:
> I tried the last build sdrangel-7.21.2-19-g12677b1bf-win64.exe but not possible to select the Antenna port, Gain,... of the associated RSPdx-r2. The buttons are present in SDRAngel but they are not active (nothing happens when clicking on them). Thanks.

Looks like the driver needed to have support added for the new SDRPLAY_RSPdxR2 device ID - which I've done in the above patch.

I'm assuming it is otherwise the same as RSPdx in terms of capabilities.

srcejon, 2024-06-17:
> I tried the last build sdrangel-7.21.2-19-g12677b1bf-win64.exe but not possible to select the Antenna port, Gain,... of the associated RSPdx-r2. 

Give this a try: https://github.com/f4exb/sdrangel/actions/runs/9548252693

ON4KHG, 2024-06-17:
I tried it and it now perfectly works ! I can select the gain, antenna port, etc
Many thanks for your efforts.
Regards,

ON4KHG

alphafox02, 2024-06-18:
I see it’s included in the Linux build, thank you! I don’t have that particular model to check, but all othered are working on 3.15. 

Should I close this out?

srcejon, 2024-06-18:
Should now work in Linux, Windows and Mac, so I think we can do.

closed 2024-06-18 by srcejon

ON4KHG, 2024-06-18:
Sure, it can be closed. Thanks.