f4exb/sdrangel #2164: USRP support missing on macos-arm

citizenrich, 2024-06-12:
Starting sdrangel on macos (from /Applications folder) on the command line shows that USRP devices are not enumerated. The error is:
```
Reason: tried: '/opt/build/sdrangel/build/external/uhd/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/uhd/lib/Python3.framework/Versions/3.9/Python3' <...>
```

srcejon, 2024-06-14:
Yep - not quite sure how to fix it though.

If you want to use USRP device on Mac, may be best to build from source yourself, so it picks up your version of python: https://github.com/f4exb/sdrangel/wiki/Compile-in-MacOS

citizenrich, 2024-06-14:
Great, thanks. I'll build from source then.

closed 2024-06-14 by citizenrich

citizenrich, 2024-06-16:
Building from source is a nightmare. 10 minutes and it invokes Java. And fails. Also, in 2024 people tend use to use homebrew not macports. Please avoid. Happy to hear others experiences on trying to build this.

robertabcd, 2025-12-28:
For people who may have the same issue on macOS:

I am able to see my USRP device on the pre-built version `7.22.9-480-g11074ac87` on macOS by starting the app from command line and passing the `UHD_IMAGES_DIR` environment variable.

For example, if your images are installed under `/opt/homebrew/Cellar/uhd/4.9.0.1_1/share/uhd/images`, then start sdrangel with this command:
```
UHD_IMAGES_DIR=/opt/homebrew/Cellar/uhd/4.9.0.1_1/share/uhd/images /Applications/SDRangel.app/Contents/MacOS/SDRangel
```