f4exb/sdrangel #2165: FileInput extremely distorted audio with WAV

Degamisu, 2024-06-12:
I am not sure how to work the FileInput with WAVs, since I don't know how to convert WAV -> SDRIQ. So I tried loading up a sample song to see it, used an AM demodulator, and it sounds very distorted. I am using a 16-bit Generic PCM file at 11.025k. the length is 3:11.406.

any help? (I am also not sure how to record the audio and send it)

srcejon, 2024-06-13:
FileInput can read .wavs directly - but it needs to be a .wav containing IQ samples. What does your .wav contain?

Degamisu, 2024-06-13:
oh. The WAV contains a japanese song, and it is not an IQ sample. How do I convert it? thats definitely the issue.

srcejon, 2024-06-14:
I'm not sure what you are trying to do.

If you want to hear audio output from AM Demodulator - then the FileInput needs to contain IQ samples of an audio signal that has been AM Modulated. To do that in SDRangel, you could use FileOutput device and the AM Modulator.

Degamisu, 2024-06-14:
> I'm not sure what you are trying to do.
> 
> If you want to hear audio output from AM Demodulator - then the FileInput needs to contain IQ samples of an audio signal that has been AM Modulated. To do that in SDRangel, you could use FileOutput device and the AM Modulator.

Oh yes. You are correct. I haven't done this before, mostly just getting NOAA data 😆 . Thanks a lot. I'll try this later.

Degamisu, 2024-06-14:
https://groups.io/g/sdrangel/topic/how_to_create_compatible_raw/29291189 This link helped a lot. It works correctly now.

closed 2024-06-14 by Degamisu