f4exb/sdrangel #2166: How to get the local output to transfer it's data to the local input?

Degamisu, 2024-06-12:
This might seem like a stupid question, but I am trying to get an ATV modulator to work and be able to encode my images into that. But I also want to read from them as well. My SDR is not working currently so im trying to simulate this as best as possible. So how do I tunnel the local output to the local input? thanks!

srcejon, 2024-06-14:
Local doesn't seem to work for me - RemoteOutput/RemoteInput does though. Just use default IP address of 127.0.0.1

srcejon, 2024-06-14:
Although there's no way to set the sample rate, as far as I can see.

Loopback support would be quite useful though.

Degamisu, 2024-06-14:
> Although there's no way to set the sample rate, as far as I can see.
> 
> Loopback support would be quite useful though.

> Although there's no way to set the sample rate, as far as I can see.
> 
> Loopback support would be quite useful though.

I could possibly just log everything to a file and then play it back from there, but it wouldn't be live unfortunately. Also, I have made a simple workspace that could possibly work, but SDRAngel wont stop crashing!

f4exb, 2024-06-17:
On this ticket the opposite direction was discussed: https://github.com/f4exb/sdrangel/issues/2082

closed 2024-07-28 by Degamisu