f4exb/sdrangel #2167: Cannot load SDRPlay library on MAC

UncleBuck62, 2024-06-15:
I just upgraded my SDRAngel to version 7.21.2 from 7.19 and not it cannot find my SDRPay RSP1A anymore.

According to the error log this is the problem:
```
2024-06-16 05:20:09.357 (D) PluginManager::loadPluginsDir: fileName: libinputsdrplayv3.dylib
2024-06-16 05:20:09.358 (W) PluginManager::loadPluginsDir: Cannot load library /Applications/SDRangel.app/Contents/Resources/lib/plugins/libinputsdrplayv3.dylib: (dlopen(/Applications/SDRangel.app/Contents/Resources/lib/plugins/libinputsdrplayv3.dylib, 0x0085): Library not loaded: libsdrplay_api.so.3
  Referenced from: <58BF731A-389C-383A-87CF-3AD28F3AA1E8> /Applications/SDRangel.app/Contents/Resources/lib/plugins/libinputsdrplayv3.dylib
  Reason: tried: 'libsdrplay_api.so.3' (no such file), '/System/Volumes/Preboot/Cryptexes/OSlibsdrplay_api.so.3' (no such file), 'libsdrplay_api.so.3' (no such file), '//libsdrplay_api.so.3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS//libsdrplay_api.so.3' (no such file), '//libsdrplay_api.so.3' (no such file))
```

Manually searching my MAC for any libsdrplay api file resulted in finding this:
`libsdrplay_api.so.3.15`
but it resides in /Library/SDRplayAPI/3.15.0/lib/libsdrplay_api.so.3.15

SDR Connect finds the RSP1A with no issues.

What can I do to resolve this problem?

System is a MAC M1 Ultra running Sonoma 14.5

srcejon, 2024-06-15:
You probably need to update to the latest SDRplay API.

UncleBuck62, 2024-06-15:
3.15 is the latest SDRPlay API

srcejon, 2024-06-16:
Ah, I forgot to update the post build step in sdrplayv3/CMakeLists.txt with the new version number. Need a better way of doing that.

srcejon, 2024-06-16:
I've uploaded a new build - give that a try.

UncleBuck62, 2024-06-16:
Sorry, no luck there either... It showed exactly the same error in the log file
```
2024-06-17 08:44:06.997 (D) PluginManager::loadPluginsDir: fileName: libinputsdrplayv3.dylib
2024-06-17 08:44:06.998 (W) PluginManager::loadPluginsDir: Cannot load library /Applications/SDRangel.app/Contents/Resources/lib/plugins/libinputsdrplayv3.dylib: (dlopen(/Applications/SDRangel.app/Contents/Resources/lib/plugins/libinputsdrplayv3.dylib, 0x0085): Library not loaded: libsdrplay_api.so.3
  Referenced from: <58BF731A-389C-383A-87CF-3AD28F3AA1E8> /Applications/SDRangel.app/Contents/Resources/lib/plugins/libinputsdrplayv3.dylib
  Reason: tried: 'libsdrplay_api.so.3' (no such file), '/System/Volumes/Preboot/Cryptexes/OSlibsdrplay_api.so.3' (no such file), 'libsdrplay_api.so.3' (no such file), '//libsdrplay_api.so.3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS//libsdrplay_api.so.3' (no such file), '//libsdrplay_api.so.3' (no such file))

```

BeppeCH, 2024-06-17:
I'm not a programmer but just wanted to share that after a few tries, the only combination with seems to be working on my MacBook Air M2 with a SDRplay RSP1B is SDRplay API 3.14.1 with SDRangel 7.20.1. Using the latest SDRangel built and/or SDRplay API 3.15 does not work, as SRangel does not recognise the device.

srcejon, 2024-06-18:
Please give the 7.21.3 release a try. There was a slight change in library naming I'd missed.

BeppeCH, 2024-06-18:
Hi srcejon. Many thanks. Version 7.21.3 works very well with SDRplay's RSP1B on the M2 MacBook Air when using SDRplay API 3.14.0. However, with API 3.15.0 SDRangel does not recognize the RSP1B. I have also access to an Intel Mac, but will only be able to test later.

BeppeCH, 2024-06-18:
I just tested the new version (7.21.3) on an Intel Mac (Mac Pro 2019 with Intel Xeon) and it works fine with the API version 3.15.0 of SDRplay. So the issue seems to be specific to Apple Silicon computers.

UncleBuck62, 2024-06-18:
Yep. I can confirm that 7.21.3 is working with SDRPlay API 3.15 now.
Thanks for all your great work.

srcejon, 2024-06-19:
> Hi srcejon. Many thanks. Version 7.21.3 works very well with SDRplay's RSP1B on the M2 MacBook Air when using SDRplay API 3.14.0. However, with API 3.15.0 SDRangel does not recognize the RSP1B. 

Not sure why that could be.

BeppeCH, 2024-06-20:
Thanks. Just to be on the safe side, I repeated the test, this time also switching off antivirus etc. with the M2 MacBook Air and came to the same results. Actually, the most stable setup is version 7.20.1 with API 3.14.1. In any case, thanks for your support!

closed 2024-08-05 by github-actions[bot]