f4exb/sdrangel #2171: Hangs when receiving signal

kenwawa01, 2024-06-17:
Hi!

I'm using SDRangel which is being recognized as

Realtek, RTL2838UHIDIR, SN: 00000001
Found Rafael Micro R828D tuner

I set it to direct sampling to approx. 100-150kHz frequencies. Every time I start using it, when generating a signal using external device and receiving it through antena, analisys stops without any visible error (acqusition stops) and if I run it from command line, I have the messages as below.

Any idea what could cause those problems?

```
2024-06-17 03:32:11.654 (I) SDRangel 7.21.1 Qt 5.15.3 64b x86_64 Ubuntu Core 22 DSP Rx:24b Tx:16b PID 57787
2024-06-17 03:32:11.654 (I) MainSettings::MainSettings: settings file: format: 0 location: /home/piotr/snap/sdrangel/9/.config/f4exb/SDRangel.conf
ALSA lib conf.c:4553:(snd_config_update_r) Cannot access file /usr/share/alsa/alsa.conf
ALSA lib conf.c:4553:(snd_config_update_r) Cannot access file /usr/share/alsa/alsa.conf
2024-06-17 03:32:11.671 (W) Unable to obtain the client patch: "org.freedesktop.DBus.Error.AccessDeniedAn AppArmor policy prevents this sender from sending this message to this recipient; type=\"method_call\", sender=\":1.1211\" (uid=1000 pid=57787 comm=\"/snap/sdrangel/9/opt/install/sdrangel/bin/sdrangel\" label=\"snap.sdrangel.sdrangel (enforce)\") interface=\"org.freedesktop.GeoClue2.Manager\" member=\"GetClient\" error name=\"(unset)\" requested_reply=\"0\" destination=\"org.freedesktop.GeoClue2\" (bus)"
2024-06-17 03:32:11.671 (W) MainCore::positionError:  0
2024-06-17 03:32:12.579 (I) PluginManager::loadPluginsDir: loaded plugin libantennatools.so
2024-06-17 03:32:12.581 (I) PluginManager::loadPluginsDir: loaded plugin libaprs.so
2024-06-17 03:32:12.581 (I) PluginManager::loadPluginsDir: loaded plugin libbeamsteeringcwmod.so
2024-06-17 03:32:12.582 (I) PluginManager::loadPluginsDir: loaded plugin libchanalyzer.so
2024-06-17 03:32:12.582 (I) PluginManager::loadPluginsDir: loaded plugin libchannelpower.so
2024-06-17 03:32:12.587 (I) PluginManager::loadPluginsDir: loaded plugin libdemodadsb.so
2024-06-17 03:32:12.588 (I) PluginManager::loadPluginsDir: loaded plugin libdemodais.so
2024-06-17 03:32:12.588 (I) PluginManager::loadPluginsDir: loaded plugin libdemodam.so
2024-06-17 03:32:12.589 (I) PluginManager::loadPluginsDir: loaded plugin libdemodapt.so
2024-06-17 03:32:12.589 (I) PluginManager::loadPluginsDir: loaded plugin libdemodatv.so
2024-06-17 03:32:12.590 (I) PluginManager::loadPluginsDir: loaded plugin libdemodbfm.so
2024-06-17 03:32:12.590 (I) PluginManager::loadPluginsDir: loaded plugin libdemodchirpchat.so
2024-06-17 03:32:12.591 (I) PluginManager::loadPluginsDir: loaded plugin libdemoddab.so
2024-06-17 03:32:12.608 (I) PluginManager::loadPluginsDir: loaded plugin libdemoddatv.so
2024-06-17 03:32:12.608 (I) PluginManager::loadPluginsDir: loaded plugin libdemoddsc.so
2024-06-17 03:32:12.609 (I) PluginManager::loadPluginsDir: loaded plugin libdemoddsd.so
2024-06-17 03:32:12.610 (I) PluginManager::loadPluginsDir: loaded plugin libdemodendoftrain.so
2024-06-17 03:32:12.610 (I) PluginManager::loadPluginsDir: loaded plugin libdemodfreedv.so
2024-06-17 03:32:12.611 (I) PluginManager::loadPluginsDir: loaded plugin libdemodft8.so
2024-06-17 03:32:12.611 (I) PluginManager::loadPluginsDir: loaded plugin libdemodils.so
2024-06-17 03:32:12.612 (I) PluginManager::loadPluginsDir: loaded plugin libdemodm17.so
2024-06-17 03:32:12.612 (I) PluginManager::loadPluginsDir: loaded plugin libdemodnavtex.so
2024-06-17 03:32:12.613 (I) PluginManager::loadPluginsDir: loaded plugin libdemodnfm.so
2024-06-17 03:32:12.613 (I) PluginManager::loadPluginsDir: loaded plugin libdemodpacket.so
2024-06-17 03:32:12.613 (I) PluginManager::loadPluginsDir: loaded plugin libdemodpager.so
2024-06-17 03:32:12.614 (I) PluginManager::loadPluginsDir: loaded plugin libdemodradiosonde.so
2024-06-17 03:32:12.614 (I) PluginManager::loadPluginsDir: loaded plugin libdemodrtty.so
2024-06-17 03:32:12.615 (I) PluginManager::loadPluginsDir: loaded plugin libdemodssb.so
2024-06-17 03:32:12.615 (I) PluginManager::loadPluginsDir: loaded plugin libdemodvor.so
2024-06-17 03:32:12.615 (I) PluginManager::loadPluginsDir: loaded plugin libdemodwfm.so
2024-06-17 03:32:12.616 (I) PluginManager::loadPluginsDir: loaded plugin libdoa2.so
2024-06-17 03:32:12.616 (I) PluginManager::loadPluginsDir: loaded plugin libfeatureafc.so
2024-06-17 03:32:12.617 (I) PluginManager::loadPluginsDir: loaded plugin libfeatureais.so
2024-06-17 03:32:12.618 (I) PluginManager::loadPluginsDir: loaded plugin libfeatureambe.so
2024-06-17 03:32:12.618 (I) PluginManager::loadPluginsDir: loaded plugin libfeaturedemodanalyzer.so
2024-06-17 03:32:12.619 (I) PluginManager::loadPluginsDir: loaded plugin libfeaturegs232controller.so
2024-06-17 03:32:12.620 (I) PluginManager::loadPluginsDir: loaded plugin libfeaturelimerfe.so
2024-06-17 03:32:12.620 (I) PluginManager::loadPluginsDir: loaded plugin libfeaturepertester.so
2024-06-17 03:32:12.620 (I) PluginManager::loadPluginsDir: loaded plugin libfeatureradiosonde.so
2024-06-17 03:32:12.621 (I) PluginManager::loadPluginsDir: loaded plugin libfeatureremotecontrol.so
2024-06-17 03:32:12.621 (I) PluginManager::loadPluginsDir: loaded plugin libfeaturerigctlserver.so
2024-06-17 03:32:12.622 (I) PluginManager::loadPluginsDir: loaded plugin libfeaturesatellitetracker.so
2024-06-17 03:32:12.623 (I) PluginManager::loadPluginsDir: loaded plugin libfeaturesid.so
2024-06-17 03:32:12.623 (I) PluginManager::loadPluginsDir: loaded plugin libfeaturesimpleptt.so
2024-06-17 03:32:12.632 (I) PluginManager::loadPluginsDir: loaded plugin libfeatureskymap.so
2024-06-17 03:32:12.632 (I) PluginManager::loadPluginsDir: loaded plugin libfeaturestartracker.so
2024-06-17 03:32:12.633 (I) PluginManager::loadPluginsDir: loaded plugin libfilesink.so
2024-06-17 03:32:12.633 (I) PluginManager::loadPluginsDir: loaded plugin libfilesource.so
2024-06-17 03:32:12.634 (I) PluginManager::loadPluginsDir: loaded plugin libfreqscanner.so
2024-06-17 03:32:12.634 (I) PluginManager::loadPluginsDir: loaded plugin libfreqtracker.so
2024-06-17 03:32:12.634 (I) PluginManager::loadPluginsDir: loaded plugin libheatmap.so
2024-06-17 03:32:12.635 (I) PluginManager::loadPluginsDir: loaded plugin libinputaaroniartsa.so
[... 124 lines cut ...]
2024-06-17 03:39:18.682 (C) RTLSDRThread: async error: Resource temporarily unavailable
2024-06-17 03:39:25.645 (C) RTLSDRInput::applySettings: could not set LO ppm correction: 0
Disabled direct sampling mode
rtlsdr_demod_read_reg failed with -1
2024-06-17 03:39:26.081 (C) RTLSDRInput::applySettings: could not set offset tuning to off
2024-06-17 03:39:26.142 (W) QObject: Cannot create children for a parent that is in a different thread.
(Parent is ChannelAPI(0x609db2b59608), parent's thread is QThread(0x609db0070900), current thread is DSPDeviceSourceEngine(0x609db10ec110)
2024-06-17 03:39:26.143 (C) SampleSinkFifo::write: () overflow - dropping 1024 samples
cb transfer status: 1, canceling...
2024-06-17 03:39:32.189 (C) RTLSDRThread: async error: Resource temporarily unavailable
2024-06-17 03:39:59.123 (C) RTLSDRInput::applySettings: could not set LO ppm correction: 0
rtlsdr_demod_read_reg failed with -7
rtlsdr_demod_write_reg failed with -9
Disabled direct sampling mode
rtlsdr_demod_read_reg failed with -7
2024-06-17 03:40:00.128 (C) RTLSDRInput::applySettings: could not set offset tuning to off
rtlsdr_demod_write_reg failed with -9
r82xx_write: i2c wr failed=-9 reg=05 len=1
2024-06-17 03:40:00.141 (C) RTLSDRInput::applySettings: error setting tuner gain mode to manual
2024-06-17 03:40:04.099 (C) RTLSDRInput::applySettings: could not set LO ppm correction: 0
Disabled direct sampling mode
2024-06-17 03:40:04.539 (C) RTLSDRInput::applySettings: could not set offset tuning to off
2024-06-17 03:40:04.599 (W) QObject: Cannot create children for a parent that is in a different thread.
(Parent is ChannelAPI(0x609db2b59608), parent's thread is QThread(0x609db0070900), current thread is DSPDeviceSourceEngine(0x609db10ec110)
```

f4exb, 2024-06-17:
Did you set up the udev rules?

srcejon, 2024-06-17:
Also, as it looks like you are using a snap - check you have enabled USB access. Either do this in Settings > Apps > SDRangel or run sudo snap connect sdrangel:raw-usb

srcejon, 2024-06-17:
The log file you sent shows that SDRangel isn't actually communicating the RTL SDR.

If Gqrx works, I guess you have udev rules installed. But check the snap permissions - or send an updated log.

kenwawa01, 2024-06-17:
Concerning usb access, I double checked it and it's granted. I've checked it in Ubuntu Software application.

Concerning udev, I'm not sure how to check it fully, but with udev monitor it seems that rules are set. Example for attaching and detaching device:

udevadm monitor
monitor will print the received events for:
UDEV - the event which udev sends out after rule processing
KERNEL - the kernel uevent

```
KERNEL[4538.196202] remove   /devices/pci0000:00/0000:00:08.1/0000:07:00.4/usb5/5-1/5-1:1.0 (usb)
KERNEL[4538.196232] remove   /devices/pci0000:00/0000:00:08.1/0000:07:00.4/usb5/5-1/5-1:1.1 (usb)
KERNEL[4538.196335] unbind   /devices/pci0000:00/0000:00:08.1/0000:07:00.4/usb5/5-1 (usb)
KERNEL[4538.196359] remove   /devices/pci0000:00/0000:00:08.1/0000:07:00.4/usb5/5-1 (usb)
UDEV  [4538.200241] remove   /devices/pci0000:00/0000:00:08.1/0000:07:00.4/usb5/5-1/5-1:1.0 (usb)
UDEV  [4538.200799] remove   /devices/pci0000:00/0000:00:08.1/0000:07:00.4/usb5/5-1/5-1:1.1 (usb)
UDEV  [4538.204550] unbind   /devices/pci0000:00/0000:00:08.1/0000:07:00.4/usb5/5-1 (usb)
UDEV  [4538.207645] remove   /devices/pci0000:00/0000:00:08.1/0000:07:00.4/usb5/5-1 (usb)
```

Currently sth changed (not knowing reason why) but I'm not even able to select this device as "Select sampling device" from SDRangel list while rtl_433 and Gqrx works without problems. Sure I run only ONE application at once.

What kind of updated log should I send as I didn't changed anything? Here is my current log:

```
2024-06-17 17:19:17.491 (I) SDRangel 7.21.1 Qt 5.15.3 64b x86_64 Ubuntu Core 22 DSP Rx:24b Tx:16b PID 18324
2024-06-17 17:19:17.491 (I) MainSettings::MainSettings: settings file: format: 0 location: /home/piotr/snap/sdrangel/9/.config/f4exb/SDRangel.conf
ALSA lib conf.c:4553:(snd_config_update_r) Cannot access file /usr/share/alsa/alsa.conf
ALSA lib conf.c:4553:(snd_config_update_r) Cannot access file /usr/share/alsa/alsa.conf
2024-06-17 17:19:17.508 (W) Unable to obtain the client patch: "org.freedesktop.DBus.Error.AccessDeniedAn AppArmor policy prevents this sender from sending this message to this recipient; type=\"method_call\", sender=\":1.450\" (uid=1000 pid=18324 comm=\"/snap/sdrangel/9/opt/install/sdrangel/bin/sdrangel\" label=\"snap.sdrangel.sdrangel (enforce)\") interface=\"org.freedesktop.GeoClue2.Manager\" member=\"GetClient\" error name=\"(unset)\" requested_reply=\"0\" destination=\"org.freedesktop.GeoClue2\" (bus)"
2024-06-17 17:19:17.509 (W) MainCore::positionError:  0
2024-06-17 17:19:18.428 (I) PluginManager::loadPluginsDir: loaded plugin libantennatools.so
2024-06-17 17:19:18.430 (I) PluginManager::loadPluginsDir: loaded plugin libaprs.so
2024-06-17 17:19:18.430 (I) PluginManager::loadPluginsDir: loaded plugin libbeamsteeringcwmod.so
2024-06-17 17:19:18.430 (I) PluginManager::loadPluginsDir: loaded plugin libchanalyzer.so
2024-06-17 17:19:18.431 (I) PluginManager::loadPluginsDir: loaded plugin libchannelpower.so
2024-06-17 17:19:18.436 (I) PluginManager::loadPluginsDir: loaded plugin libdemodadsb.so
2024-06-17 17:19:18.436 (I) PluginManager::loadPluginsDir: loaded plugin libdemodais.so
2024-06-17 17:19:18.437 (I) PluginManager::loadPluginsDir: loaded plugin libdemodam.so
2024-06-17 17:19:18.437 (I) PluginManager::loadPluginsDir: loaded plugin libdemodapt.so
2024-06-17 17:19:18.438 (I) PluginManager::loadPluginsDir: loaded plugin libdemodatv.so
2024-06-17 17:19:18.438 (I) PluginManager::loadPluginsDir: loaded plugin libdemodbfm.so
2024-06-17 17:19:18.439 (I) PluginManager::loadPluginsDir: loaded plugin libdemodchirpchat.so
2024-06-17 17:19:18.440 (I) PluginManager::loadPluginsDir: loaded plugin libdemoddab.so
2024-06-17 17:19:18.457 (I) PluginManager::loadPluginsDir: loaded plugin libdemoddatv.so
2024-06-17 17:19:18.457 (I) PluginManager::loadPluginsDir: loaded plugin libdemoddsc.so
2024-06-17 17:19:18.458 (I) PluginManager::loadPluginsDir: loaded plugin libdemoddsd.so
2024-06-17 17:19:18.458 (I) PluginManager::loadPluginsDir: loaded plugin libdemodendoftrain.so
2024-06-17 17:19:18.459 (I) PluginManager::loadPluginsDir: loaded plugin libdemodfreedv.so
2024-06-17 17:19:18.459 (I) PluginManager::loadPluginsDir: loaded plugin libdemodft8.so
2024-06-17 17:19:18.460 (I) PluginManager::loadPluginsDir: loaded plugin libdemodils.so
2024-06-17 17:19:18.460 (I) PluginManager::loadPluginsDir: loaded plugin libdemodm17.so
2024-06-17 17:19:18.461 (I) PluginManager::loadPluginsDir: loaded plugin libdemodnavtex.so
2024-06-17 17:19:18.461 (I) PluginManager::loadPluginsDir: loaded plugin libdemodnfm.so
2024-06-17 17:19:18.462 (I) PluginManager::loadPluginsDir: loaded plugin libdemodpacket.so
2024-06-17 17:19:18.462 (I) PluginManager::loadPluginsDir: loaded plugin libdemodpager.so
2024-06-17 17:19:18.463 (I) PluginManager::loadPluginsDir: loaded plugin libdemodradiosonde.so
2024-06-17 17:19:18.463 (I) PluginManager::loadPluginsDir: loaded plugin libdemodrtty.so
2024-06-17 17:19:18.464 (I) PluginManager::loadPluginsDir: loaded plugin libdemodssb.so
2024-06-17 17:19:18.464 (I) PluginManager::loadPluginsDir: loaded plugin libdemodvor.so
2024-06-17 17:19:18.464 (I) PluginManager::loadPluginsDir: loaded plugin libdemodwfm.so
2024-06-17 17:19:18.465 (I) PluginManager::loadPluginsDir: loaded plugin libdoa2.so
2024-06-17 17:19:18.465 (I) PluginManager::loadPluginsDir: loaded plugin libfeatureafc.so
2024-06-17 17:19:18.466 (I) PluginManager::loadPluginsDir: loaded plugin libfeatureais.so
[... cut ...] \" member=\"GetDevices\" error name=\"(unset)\" requested_reply=\"0\" destination=\"org.freedesktop.NetworkManager\" (uid=0 pid=1937 comm=\"/usr/sbin/NetworkManager --no-daemon \" label=\"unconfined\")"
2024-06-17 17:19:19.633 (W) "Object path cannot be empty"
2024-06-17 17:19:19.667 (W) libpng warning: iCCP: known incorrect sRGB profile
2024-06-17 17:19:19.667 (W) libpng warning: iCCP: too many profiles
2024-06-17 17:19:19.692 (I) WebAPIServer::start: starting web API server at http://:8091
```

srcejon, 2024-06-17:
Well, the first log showed there was a problem communicating with the RTL SDR, which is why you couldn't see anything.

The second log looks OK, so you should be seeing something.

kenwawa01, 2024-06-17:
So it's exactly opposite. In the first case I was able to see noise but signal was causing software to hang. In the second, I was not able even to connect to the device...

closed 2024-08-02 by github-actions[bot]