f4exb/sdrangel #2179: SDRAngel Crashing on open

farm24, 2024-06-20:
With the newest maintenance update, whenever i try to open sdrangel it just crashes when loading plugins shows up in the loading bar. I've tried reinstalling, restarting, trying the Github Actions releases, and nothing helps.

srcejon, 2024-06-20:
What os?

Run you run from the command line so we can see any debug output?

You could try running with --scratch option, to start with an empty workspace.

Also try starting with SDRs unplugged

farm24, 2024-06-20:
im running windows and have no SDR's plugged in

farm24, 2024-06-20:
and running it with --scratch in the command line does nothing and doesnt return anything either

f-alex, 2024-06-20:
Has the same problem. Win 11 logs point that problem in this lib MSVCP140.dll. Download and inslall latest visual C++ redistributable pack and all works
https://aka.ms/vs/17/release/vc_redist.x64.exe
or x86)

srcejon, 2024-06-21:
> Has the same problem. Win 11 logs point that problem in this lib MSVCP140.dll. Download and inslall latest visual C++ redistributable pack and all works https://aka.ms/vs/17/release/vc_redist.x64.exe or x86)

This appears to be version 14.40.33810. 

We are currently including 14.30.30708: https://github.com/f4exb/sdrangel-windows-libraries/commits/master/vcredist

srcejon, 2024-06-21:
Have created a PR in sdrangel-windows-libraries with the latest VC Redistributables: https://github.com/f4exb/sdrangel-windows-libraries/pull/24

Probably worth updating regardless.

PR #2185 (2024-06-26): Update sdrangel-windows-libraries link 
Update sdrangel-windows-libraries link to include latest VC redistributable. Hopefully fixes #2179
closed 2024-06-26 by f4exb