f4exb/sdrangel #2183: Autosave workspaces

Degamisu, 2024-06-23:
This is very important for me, since SDRangel will crash for any reason. It will reset to the configuration when it was last closed, and I lose all the configurations that i made after that. Autosaving the workspace will help with that a TON!

f4exb, 2024-06-23:
This would be very detrimental to performance. You can save all presets, current configuration etc... with the Preferences > Save all command.

Degamisu, 2024-06-23:
> This would be very detrimental to performance. You can save all presets, current configuration etc... with the Preferences > Save all command.

what sort of performance issues would there be if its saved in intervals? i would understand that performance impact though, this software already uses enough cpu 😆

dforsi, 2024-06-23:
To change you complaint into an actionable problem report you need to be more specific on what doesn't work in your use case.
What doesn't work for you in editing a configuration and saving it when you are happy with it?
Sdrangel shouldn't crash but just saving in intervals will destroy many disks.

Degamisu, 2024-06-23:
> To change you complaint into an actionable problem report you need to be more specific on what doesn't work in your use case. What doesn't work for you in editing a configuration and saving it when you are happy with it? Sdrangel shouldn't crash but just saving in intervals will destroy many disks.

oh! i had no idea it would break disks. makes sense though. its hard to say the actual problem in this case however, as it randomly crashes at any time, like scrolling through sampling devices with the middle mouse, to adding an AM demodulator. Ill close this as "wontfix"

closed 2024-06-23 by Degamisu