f4exb/sdrangel #2189: SDRAngel Android: Error saving files

rkarikari, 2024-06-28:
Hello,

I'm unable to save any files in sdrangel android. 
Any attempt results in a zero-byte file and an error message as shown in the attached screenshots.

Any help / suggestions would be greatly appreciated.

Thanks.

[image]
[image]

srcejon, 2024-06-28:
Android needs special file handling, and that has yet been implemented yet for exporting configuration files.

rkarikari, 2024-06-28:
Thanks for the prompt response.

I guess the same issue applies to saving anything in sdrangel for android (downloaded/decoded APT images) as I got the same error when trying to save.

srcejon, 2024-06-29:
Saving images from APT demod works for me.

srcejon, 2024-06-29:
Actually, saving the configuration should work as well, but you need to make sure you enter a filename with the .cfgx extension.

Likewise, when saving APT image, make sure you enter a .jpg or .png extension in the file dialog.

(Android wont let SDRangel add the extension itself).

rkarikari, 2024-06-29:
> Actually, saving the configuration should work as well, but you need to make sure you enter a filename with the .cfgx extension.
> 
> Likewise, when saving APT image, make sure you enter a .jpg or .png extension in the file dialog.
> 
> (Android wont let SDRangel add the extension itself).

I tried  the .jpg extention and got the following error...
[image]

However, using .png works !
I've also  confirmed that saving works for the configuration files once the .cfgx extension is appended to the file name.

Thanks so much.

closed 2024-07-04 by srcejon