f4exb/sdrangel #2191: Hey It NOT ISSUES But i need colormap Add cubehelix

KristapsSDRSharp, 2024-06-30:
[image]

f4exb, 2024-06-30:
It is pretty difficult to retrieve values from a picture. In SDRangel color maps are stored as 256 (R,G,B) floating point values (0.0 to 1.0) from lower to higher. If you can provide these values it will be no big deal to add this color map.

f4exb, 2024-07-25:
This can be done in Python...
Firstly install pillow:
```
pip install pillow
```
Then this snippet should print out the colormap structure to be inserted in `sdrgui\gui\colormap.cpp`:
```python
from PIL import Image

im = Image.open('cubehelix.png') 
pix = im.load()
xsize = im.size[0]

print("const float ColorMap::m_cubehelix[m_size] =")
print("{")

for xi in range(256):
    x = int((xsize/256)*xi)
    colortuple = pix[x, 0][:-1]
    rf = colortuple[0] / 256
    gf = colortuple[1] / 256
    bf = colortuple[2] / 256
    print(f"    {rf:.4f},   {gf:.4f},   {bf:.4f},")

print("};")
```

f4exb, 2024-07-25:
The explanation that goes with this palette seems to be here: https://people.phy.cam.ac.uk/dag9/CUBEHELIX/ Could be nice indeed.

f4exb, 2024-07-26:
Two new colormaps:

- cubehelix: derived from the attached image which has linear grayscale progression
- cubegamma: calculated with the javascript app applying a gamma factor of 0.4 to accentuate weak signals

commit 78b73ae5 (2024-07-26): Added cubehelix and cubegamma color maps. Default to angel if not found. Fixes #2191
closed 2024-07-26 by f4exb