f4exb/sdrangel #2192: qtgamepad not exist in QT6

hiperiondev, 2024-06-30:
I am the maintainer of the Sdrangel build for fedora.
qtgamepad does not exist for QT6 and it is very possible that it will not be ported for quite some time (the developer wants to make many changes before making the port)
Is there any alternative to its use? Is it still necessary?

f4exb, 2024-06-30:
As far as I can see this is used only in the Rotor controller feature (gs232controller) so you can always skip this plugin from your build. However maybe you do want this feature.

hiperiondev, 2024-06-30:
Thank you very much for the reply. It does not seem to generate conflicts in operation using the qt5 library

hiperiondev, 2024-06-30:
But when it starts I find the following error:

2024-06-30 19:14:06.248 (W) AudioOutputDevice::start: using system default device Default ALSA Output (currently PipeWire Media Server)
2024-06-30 19:14:06.248 (I) AudioOutputDevice::start: audio format OK
ALSA lib pcm.c:8750:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8750:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8750:(snd_pcm_recover) underrun occurred

srcejon, 2024-07-01:
> But when it starts I find the following error:
>
> 2024-06-30 19:14:06.248 (W) AudioOutputDevice::start: using system default device Default ALSA Output (currently PipeWire Media Server) 2024-06-30 [...]

This is unrelated to qtgamepad.

srcejon, 2024-07-01:
> As far as I can see this is used only in the Rotor controller feature (gs232controller) so you can always skip this plugin from your build. However maybe you do want this feature.

The plugin should still compile and work even without qtgamepad - the code that uses it should all be conditional on if it is available. If it doesn't compile, please let me know.

hiperiondev, 2024-07-05:
Yes, compile and work.
Txs

closed 2024-07-05 by hiperiondev