f4exb/sdrangel #2194: 7.21.4 fails to open W11 x64.

mistergitj, 2024-07-03:
sdrangel-7.21.4-win64 fails to open on W11 x64. Event Viewer:
+ System 

  - Provider 

   [ Name]  Application Error 
   [ Guid]  {a0e9b465-b939-57d7-b27d-95d8e925ff57} 
 
   EventID 1000 
 
   Version 0 
 
   Level 2 
 
   Task 100 
 
   Opcode 0 
 
   Keywords 0x8000000000000000 
 
  - TimeCreated 

   [ SystemTime]  2024-07-03T15:35:58.2571577Z 
 
   EventRecordID 13083 
 
   Correlation 
 
  - Execution 

   [ ProcessID]  12540 
   [ ThreadID]  15984 
 
   Channel Application 
 
   Computer xxxxxxxxxxxxxxxxxxxx
 
  - Security 

   [ UserID]  S-1-5-21-486078230-4218467414-859634492-1000 
 

- EventData 

  AppName sdrangel.exe 
  AppVersion 0.0.0.0 
  AppTimeStamp 667b2fd6 
  ModuleName MSVCP140.dll 
  ModuleVersion 14.34.31938.0 
  ModuleTimeStamp e7a4b09e 
  ExceptionCode c0000005 
  FaultingOffset 0000000000013028 
  ProcessId 0x44c8 
  ProcessCreationTime 0x1dacd5eb25e4693 
  AppPath C:\Program Files\SDRangel\sdrangel.exe 
  ModulePath C:\WINDOWS\SYSTEM32\MSVCP140.dll 
  IntegratorReportId f0fcfb69-6992-40e2-93b5-2d71b199777e 
  PackageFullName  
  PackageRelativeAppId

srcejon, 2024-07-03:
Try installing the latest Visual C libraries from: https://aka.ms/vs/17/release/vc_redist.x64.exe

mistergitj, 2024-07-03:
Thanks much, srcejon, fixed. John.

closed 2024-07-04 by srcejon