f4exb/sdrangel #2196: SDRAngel stutters 30 seconds after being put to background on Android

runbert, 2024-07-04:
Hi!

I've set up SDRAngel on my Android Galaxy S8 phone, and almost everything works great with R820t2.

The thing is, if I tune in a WFM radio station and use the WFM modulator to listen to it, the station sounds great when the app is in foreground. However, if I put the app to background or turn off the screen, the current received radio station starts to stutter exactly 30 seconds after it has been put to background.

I've turned all power saving features off, plugged the phone into a charger etc. but it still starts to stutter after 30 seconds in background. I even factory resetted the device and made a fresh installation of android without any luck.

I've also tried with NFM demodulator but still the same.

It would be so nice to be able to have the screen off to avoid LCD burn in and save power...

Anyone knows what might be wrong or some things to try?

Thanks!

srcejon, 2024-07-04:
I suspect we may need a change that tells Android to allow it to run properly when in the background. What version of Android do you have?

runbert, 2024-07-04:
> I suspect we may need a change that tells Android to allow it to run properly when in the background. What version of Android do you have?

Yes. That’s what I think too. I’m running the latest version that the Galaxy S8 supports which is Android 9