f4exb/sdrangel #2197: SDRAngel stops receiving after 7 hours on Android

runbert, 2024-07-05:
Hi,

Im using a generic R820t2 dongle and run sdrangel on android in foreground with screen on. The thing is, sdrangel loses connection to the dongle after around 7 hours and its not working to press the stop/play button to restart it. Only way to get it working again is to restart sdrangel and then press play again.

Running on a newly installed Samsung galaxy S8.

Any ideas of what might be wrong and what to try?

Thanks!

closed 2024-07-05 by runbert

srcejon, 2024-07-05:
Is the dongle getting hot?

runbert, 2024-07-05:
> Is the dongle getting hot?

I'll check next time it fails. But I think it gets pretty hot, yes.

I recently discovered that SDRAngel loses connection to the SDR-dongle and has to be restarted whenever my AC adapter plugs in or out, so it could be related to that issue to. The AC adapter maybe disconnects for a split second and so and therefore make SDR Angel to stop working.