f4exb/sdrangel #2199: Cannot load library & Could not start sample source

CuproAT, 2024-07-05:
Hello

I try to use SDR Angel with HackRF and a MacBook M2 and on Windows 11. But here only the MacOS way.
I get the "sample source" error which was mentioned also in other issues and also "cannot load library" in the attached log.

MacOS Sonoma 14.5

Found HackRF
Board ID Number: 2 (HackRF One)
Firmware Version: 2024.02.1 (API:1.08)
Part ID Number: 0xa000cb3c 0x0053475b
Hardware Revision: older than r6

[sdrangel.txt](https://github.com/user-attachments/files/16110246/sdrangel.txt)

CocolinoFan, 2024-07-12:
Do you have installed:
`brew install hackrf` (Homebrew) or `sudo port install hackrf` (MacPorts) for your Mac?
On Windows I think the hackrf library can be installed from [here](https://downloads.myriadrf.org/builds/PothosSDR/)

CocolinoFan, 2024-07-12:
```
2024-07-05 15:17:27.164 (C) DeviceHackRF::open_hackrf: error #-5: HackRF not found
2024-07-05 15:17:27.164 (C) HackRFInput::openDevice: could not open HackRF 14d463dcf3d4be1
```
Hmmm, do you use the same cable for connecting the HackRF to the Mac and to Windows? Are you sure both the Mac and Windows have **USB 3** ?

CuproAT, 2024-07-12:
gqrx is working on my MacBook, so homebrew hackrf is installed

yes, USB-C to MicroUSB

closed 2024-08-27 by github-actions[bot]