f4exb/sdrangel #2201: WFM modulator enhancements

Degamisu, 2024-07-06:
Some things I would love to have in the WFM modulator would be RBDS modulation and louder volume/power. Ive noticed that some of my broadcasts have a low power (around -60db) which is not ideal. The RBDS modulation will also help with adding context to the broadcast!

f4exb, 2024-07-06:
The WFM modulator is basically a FM modulator with a possibility of a wider excursion than NFM. This is not a broadcast FM modulator which in addition to RDS would require stereo support. In fact you need the stereo pilot for RDS also. In fact this should be implemented as a new plugin.

Degamisu, 2024-07-06:
> The WFM modulator is basically a FM modulator with a possibility of a wider excursion than NFM. This is not a broadcast FM modulator which in addition to RDS would require stereo support. In fact you need the stereo pilot for RDS also. In fact this should be implemented as a new plugin.

Sounds good!

jfdelnero, 2024-07-14:
I have developed an experimental broadcast Stereo FM + RDS modulator some time ago.

https://github.com/jfdelnero/rf-tools/blob/master/src/broadcast_fm/broadcast_fm.c

Here is the current modulator design :

```
//
// Broadcast FM subcarriers
//
//      ________            Stereo L - R modulation
//     /        \                38KHz DSB-SC
//    /          \           _______      _______
//   /    MONO    \  Pilot  /       \    /       \     RDS (57KHz DSB-SC)
//  /     L + R    \   |   /         \  /         \    _  _
// /                \__|__/           \/           \__/ \/ \___
//30Hz           15Khz |  |            |            |    |
//                     |  |            |            |  57KHz 5%
//                  19KHz 23KHz       38KHz         53KHz
//                    10%
//

// ---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
// Current software design :
//
// Mod player  /  Left Audio  -> Preemphasis (FIR) -> \  / -> Left + Right -> 15KHz low pass (FIR) >------------------------------|
// Sound                                               \/                                                                         |
// generator                                           /\                                                                         |
//             \  Right Audio -> Preemphasis (FIR) -> /  \ -> Left - Right -> 15KHz low pass (FIR)                                |
//                                                                                                |                               |
// Sample      |  Sample Rate at 200KHz                                                           V                               |
// Rate at     |                                                                                |MUL| (38KHz DSB-SC modulated)->  + ----> I+Q Modulator --> RF hardware transceiver
// 50KHz       |                                                                                  ^                               |     (2MHz sample rate)
//             |                                                                                  |                               |
//             |                                                                               38KHz Osc                          |
//             |                                                                                 (|) (These oscillators           |
//             |                                                                                 (|)  must be kept in phase)      |
//             |                                                                               19KHz Pilot Osc  >-----------------+
//             |                                                                                 (|)                              |
//             |                                                                                 (|)      (57KHz DSB-SC modulated)|
//             |                                                                               57KHz Osc-->|MUL|>-----------------+
//                                                                                                |          ^
//                                                                                           ____ v_____     |
//                                                                                          |           |    |
//                                                                                          |    RDS    |>----
//                                                                                          | Generator |
//                                                                                          |___________|
//
//
// ---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
```

Degamisu, 2024-07-15:
Excelent work! I'll test this out as soon as possible.

iibaranov-IG, 2026-09-09:
Мы готовы помочь с RDS/RBDS как отдельным первым этапом WFM-модуля. Предлагаем узкий scope:

- PI-код, PS (8 символов), PTY/TP/TA и RadioText;
- корректное формирование RDS-поднесущей 57 kHz с заданной частотой символов и CRC;
- параметры в WebAPI/GUI с безопасными значениями по умолчанию;
- unit-тесты для  групп, differential encoding и CRC;
- без попытки «увеличить RF power» программно — это зависит от SDR/усилителя.

Перед реализацией нужно подтвердить, какой SDR/самотест автор использует и какой формат API WFMMod предпочтителен. Готовы подготовить прототип encoder + тестовый вектор, а эфирную проверку провести на предоставленном стенде или IQ-capture.

srcejon, 2026-10-27:
Implemented in v8

closed 2026-10-27 by srcejon