f4exb/sdrangel #2203: ADSB, no map display

bazineta, 2024-07-08:
Using the compiled ARM Mac binary, maps other than the street maps do not display within the ADS-B demodulator. The following appears in a debug trace:
```
2024-07-06 14:02:41.778 (W) qrc:/map/map_6.qml:66:17 Parameter "mouse" is not declared. Injection of parameters into signal handlers is deprecated. Use JavaScript functions with formal parameters instead.
2024-07-06 14:02:41.778 (W) qrc:/map/map_6.qml:316:29 Parameter "mouse" is not declared. Injection of parameters into signal handlers is deprecated. Use JavaScript functions with formal parameters instead.
2024-07-06 14:03:16.129 (W) QGeoTileRequestManager: Failed to fetch tile (9,25,6) 5 times, giving up. Last error message was: 'Error transferring https://maps.wikimedia.org/osm-intl/6/9/25.png - server replied: '
2024-07-06 14:03:16.145 (W) QGeoTileRequestManager: Failed to fetch tile (9,24,6) 5 times, giving up. Last error message was: 'Error transferring https://maps.wikimedia.org/osm-intl/6/9/24.png - server replied: '
```
When I manually test the URLs listed, Wikimedia issues the following error:
```
Error: 403, Forbidden: Map tiles are restricted to Wikimedia & affiliated sites only. See https://wikitech.wikimedia.org/wiki/Maps/External_usage if you believe your usage supports the Movement. 
```
Maps within the Map feature do work correctly; it's just maps other than street maps in the ADS-B demodulator that do not.

srcejon, 2025-01-25:
> Using the compiled ARM Mac binary, 7.21.4, maps other than the street maps do not display within the ADS-B demodulator. 

Aviation and satellite maps work for me on 7.22.5 on mac arm. Can you try the latest release please?

bazineta, 2025-01-25:
Working properly with 7.22.5.

closed 2025-01-25 by srcejon