f4exb/sdrangel #2204: Build fails with `datvideorender.cpp:282:46: error: ‘AVCodecContext’ {aka ‘struct AVCodecContext’} has no member named ‘channels’`; and warning `AutoUic: /var/cache/makepkg/build/sdrangel-git/src/sdrangel/plugins/channelrx/demodft8/ft8demodgui.ui: Warning: The name 'verticalLayout_2' (QVBoxLayout) is already in use, defaulting to 'verticalLayout_21'`.

dreirund, 2024-07-08:
I build the latst state from this git repository according to [this recipe](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=sdrangel-git) on [Artix GNU/Linux](http://artixlinux.org/), GCC version: 14.1.1.

Build fails for me with `datvideorender.cpp:282:46: error: ‘AVCodecContext’ {aka ‘struct AVCodecContext’} has no member named ‘channels’`:  
```
[...]
[ 40%] Automatic MOC and UIC for target demoddatv
[ 40%] Built target demoddatv_autogen
[ 40%] Building CXX object plugins/channelrx/demoddatv/CMakeFiles/demoddatv.dir/demoddatv_autogen/mocs_compilation.cpp.o
[ 40%] Building CXX object plugins/channelrx/demoddatv/CMakeFiles/demoddatv.dir/datvdemod.cpp.o
[ 40%] Building CXX object plugins/channelrx/demoddatv/CMakeFiles/demoddatv.dir/datvdemodgui.cpp.o
[ 40%] Building CXX object plugins/channelrx/demoddatv/CMakeFiles/demoddatv.dir/datvdemodplugin.cpp.o
[ 40%] Building CXX object plugins/channelrx/demoddatv/CMakeFiles/demoddatv.dir/datvdemodsettings.cpp.o
[ 40%] Building CXX object plugins/channelrx/demoddatv/CMakeFiles/demoddatv.dir/datvdemodwebapiadapter.cpp.o
[ 40%] Building CXX object plugins/channelrx/demoddatv/CMakeFiles/demoddatv.dir/datvideostream.cpp.o
[ 40%] Building CXX object plugins/channelrx/demoddatv/CMakeFiles/demoddatv.dir/datvudpstream.cpp.o
[ 40%] Building CXX object plugins/channelrx/demoddatv/CMakeFiles/demoddatv.dir/datvideorender.cpp.o
/var/cache/makepkg/build/sdrangel-git/src/sdrangel/plugins/channelrx/demoddatv/datvideorender.cpp: In member function ‘bool DATVideoRender::preprocessStream()’:
/var/cache/makepkg/build/sdrangel-git/src/sdrangel/plugins/channelrx/demoddatv/datvideorender.cpp:282:46: error: ‘AVCodecContext’ {aka ‘struct AVCodecContext’} has no member named ‘channels’
  282 |     m_metaData.Channels = m_videoDecoderCtx->channels;
      |                                              ^~~~~~~~
/var/cache/makepkg/build/sdrangel-git/src/sdrangel/plugins/channelrx/demoddatv/datvideorender.cpp:308:48: error: ‘AVCodecContext’ {aka ‘struct AVCodecContext’} has no member named ‘channels’
  308 |         << " channels: " << m_audioDecoderCtx->channels
      |                                                ^~~~~~~~
/var/cache/makepkg/build/sdrangel-git/src/sdrangel/plugins/channelrx/demoddatv/datvideorender.cpp:309:54: error: ‘AVCodecContext’ {aka ‘struct AVCodecContext’} has no member named ‘channel_layout’; did you mean ‘ch_layout’?
  309 |         << " channel_layout: " << m_audioDecoderCtx->channel_layout
      |                                                      ^~~~~~~~~~~~~~
      |                                                      ch_layout
/var/cache/makepkg/build/sdrangel-git/src/sdrangel/plugins/channelrx/demoddatv/datvideorender.cpp: In member function ‘void DATVideoRender::setResampler()’:
/var/cache/makepkg/build/sdrangel-git/src/sdrangel/plugins/channelrx/demoddatv/datvideorender.cpp:628:72: error: ‘AVCodecContext’ {aka ‘struct AVCodecContext’} has no member named ‘channels’
  628 |     av_opt_set_int(m_audioSWR, "in_channel_count",  m_audioDecoderCtx->channels, 0);
      |                                                                        ^~~~~~~~
/var/cache/makepkg/build/sdrangel-git/src/sdrangel/plugins/channelrx/demoddatv/datvideorender.cpp:630:73: error: ‘AVCodecContext’ {aka ‘struct AVCodecContext’} has no member named ‘channel_layout’; did you mean ‘ch_layout’?
  630 |     av_opt_set_int(m_audioSWR, "in_channel_layout",  m_audioDecoderCtx->channel_layout, 0);
      |                                                                         ^~~~~~~~~~~~~~
      |                                                                         ch_layout
/var/cache/makepkg/build/sdrangel-git/src/sdrangel/plugins/channelrx/demoddatv/datvideorender.cpp:640:57: error: ‘AVCodecContext’ {aka ‘struct AVCodecContext’} has no member named ‘channels’
  640 |         << " in_channel_count: " <<  m_audioDecoderCtx->channels
      |                                                         ^~~~~~~~
/var/cache/makepkg/build/sdrangel-git/src/sdrangel/plugins/channelrx/demoddatv/datvideorender.cpp:642:58: error: ‘AVCodecContext’ {aka ‘struct AVCodecContext’} has no member named ‘channel_layout’; did you mean ‘ch_layout’?
  642 |         << " in_channel_layout: " <<  m_audioDecoderCtx->channel_layout
      |                                                          ^~~~~~~~~~~~~~
      |                                                          ch_layout
At global scope:
cc1plus: note: unrecognized command-line option ‘-Wno-inconsistent-missing-override’ may have been intended to silence earlier diagnostics
make[2]: *** [plugins/channelrx/demoddatv/CMakeFiles/demoddatv.dir/build.make:195: plugins/channelrx/demoddatv/CMakeFiles/demoddatv.dir/datvideorender.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:11471: plugins/channelrx/demoddatv/CMakeFiles/demoddatv.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
make: Leaving directory '/var/cache/makepkg/build/sdrangel-git/src/build'
```

The `$CXXFLAGS` in use are  
`-march=native -mtune=native -O3 -pipe -fno-plt -fexceptions -Wp,-D_FORTIFY_SOURCE=2 -fstack-clash-protection -fcf-protection -w -Wp,-D_GLIBCXX_ASSERTIONS -fpermissive`.

Output of the `cmake` call (`cmake` command line arguments: `-Wno-dev -DARCH_OPT="" -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr -DLIBDSDCC_INCLUDE_DIR=/usr/include/dsdcc -DCM256CC_INCLUDE_DIR=/usr/include/cm256cc`):

```
-- The C compiler identification is GNU 14.1.1
-- The CXX compiler identification is GNU 14.1.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.45.2")
-- Using ccache /usr/bin/ccache
[... 146 lines cut ...]
-- Asking qmake for QT_PLUGINS_DIR and got /usr/lib/qt/plugins
-- Asking qmake for QT_IMPORTS_DIR and got /usr/lib/qt/imports
-- Asking qmake for QT_QML_DIR and got /usr/lib/qt/qml
-- Asking qmake for QT_LIBS_DIR and got /usr/lib
-- CPack generators: 
-- Configuring done (2.6s)
-- Generating done (0.9s)
-- Build files have been written to: /var/cache/makepkg/build/sdrangel-git/src/build
```

---

Also note that I get the following warning during the build process:

```
[...]
[ 38%] Built target demodft8_autogen_timestamp_deps
[ 38%] Automatic MOC and UIC for target demodft8
AutoUic: /var/cache/makepkg/build/sdrangel-git/src/sdrangel/plugins/channelrx/demodft8/ft8demodgui.ui: Warning: The name 'verticalLayout_2' (QVBoxLayout) is already in use, defaulting to 'verticalLayout_21'.
[...]
```

---

Regards!

srcejon, 2024-07-17:
The latest version of ffmpeg isn't currently supported.

calimeroteknik, 2024-07-19:
The question to resolve this is : which is the (recommended) way to specify the path for ffmpeg 4.4?

I tried messing around in cmake parameters and editing a bunch of files with ffmpeg paths (CMakeLists.txt mostly), but so far to no avail; I don't understand the build system.
On Arch Linux, ffmpeg4.4 installs to `/usr/include/ffmpeg4.4/` and `/usr/lib/ffmpeg4.4/` instead of `/usr/include` and `/usr/lib`.

How can I make it so the build message,
```
-- Found FFmpeg: /usr/lib/libavcodec.so;/usr/lib/libavformat.so;/usr/lib/libavutil.so;/usr/lib/libswscale.so
```
changes to this?
```
-- Found FFmpeg: /usr/lib/ffmpeg4.4/libavcodec.so;/usr/lib/ffmpeg4.4/libavformat.so;/usr/lib/ffmpeg4.4/libavutil.so;/usr/lib/ffmpeg4.4/libswscale.so
```

calimeroteknik, 2024-07-20:
Running cmake from `strace -f` with the option `-DPKG_CONFIG_ARGN="--with-path=/usr/lib/ffmpeg4.4/pkgconfig"` it turns out that:

1. The argument is indeed supplied to a pkg-config process, after the cmake process has read `cmake/Modules/FindFFmpeg.cmake`:
```
read(4, "# vim: ts=2 sw=2\n# - Try to find the required ffmpeg components(default: AVFORMAT, AVUTIL, AVCODEC)\n#\n# Once done this will define\n#  FFMPEG_FOUND         - System has the all required com"..., 8192) = 5643
write(1, "-- Checking for module 'libavcodec'\n", 36) = 36
[pid 2265598] execve("/usr/bin/pkg-config", ["/usr/bin/pkg-config", "--with-path=/usr/lib/ffmpeg4.4/pkgconfig", "--print-errors", "--short-errors", "--exists", "libavcodec"], 0x5cd73a08ec70 /* 92 vars */ <unfinished ...>
[pid 2265598] openat(AT_FDCWD, "/usr/lib/ffmpeg4.4/pkgconfig/libavcodec.pc", O_RDONLY) = 4
[pid 2265598] read(4, "prefix=/usr\nexec_prefix=${prefix}\nlibdir=/usr/lib/ffmpeg4.4\nincludedir=/usr/include/ffmpeg4.4\n\nName: libavcodec\nDescription: FFmpeg codec library\nVersion: 58.134.100\nRequires: \nRequires.private: libswresample >= 3.9.100, libavutil >= 56.70.100\nConflicts:\nLibs: -L${libdir}  -lavcodec \nLibs.private: -lvpx -lm -lvpx -lm -lvpx -lm -lvpx -lm -lwebpmux -pthread -lm -latomic -llzma -ldav1d -lopencore-amrwb -lrsvg-2 -lm -lgio-2.0 -lglib-2.0 -lgdk_pixbuf-2.0 -lgobject-2.0 -lcairo -lz -laom -lgsm -lmp3lame -lm -lopencore-amrnb -lopenjp2 -lopus -lrav1e -lspeex -lSvtAv1Enc -ltheoraenc -ltheoradec -logg -lvorbis -lvorbisenc -lwebp -lx264 -lx265 -lxvidcore -lva -lmfx -lstdc++ -ldl\nCflags: -I${includedir}\n", 4096) = 700
```
2. The cmake process announces a success:
```
write(1, "--   Found libavcodec, version 58.134.100\n", 42) = 42
```
3. The library is even looked at by cmake:
```
access("/usr/lib/ffmpeg4.4/libavcodec.so", R_OK) = 0
```
4. Erm, and then cmake looks for includes everywhere in `$PATH` and more, but (this is a problem) **never** in the place specified by `/usr/lib/ffmpeg4.4/pkgconfig/libavcodec.pc` (which contains `includedir=/usr/include/ffmpeg4.4`, and `/usr/include/ffmpeg4.4/libavcodec/avcodec.h` exists where it should be):
 ```
access("/usr/lib/nx/bin/ffmpeg/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/usr/lib/nx/bin/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/usr/local/sbin/ffmpeg/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/usr/local/sbin/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/usr/local/bin/ffmpeg/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/usr/local/bin/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/usr/bin/ffmpeg/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/usr/bin/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/usr/lib/jvm/default/bin/ffmpeg/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/usr/lib/jvm/default/bin/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/usr/bin/site_perl/ffmpeg/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/usr/bin/site_perl/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/usr/bin/vendor_perl/ffmpeg/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/usr/bin/vendor_perl/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/usr/bin/core_perl/ffmpeg/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/usr/bin/core_perl/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/opt/qt/bin/ffmpeg/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/opt/qt/bin/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/home/calimero/.bin/ffmpeg/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/home/calimero/.bin/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/usr/local/include/ffmpeg/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/usr/local/include/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/usr/local/ffmpeg/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/usr/local/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/usr/include/ffmpeg/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/usr/include/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/usr/ffmpeg/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/usr/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/include/ffmpeg/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/include/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/usr/X11R6/include/ffmpeg/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/usr/X11R6/include/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/usr/X11R6/ffmpeg/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/usr/X11R6/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
access("/usr/pkg/include/ffmpeg/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
[... cut ...]
access("/usr/include/X11/libavcodec/avcodec.h", R_OK) = -1 ENOENT (No such file or directory)
```
5. Finally sdrangel's `cmake/Modules/FindFFmpeg.cmake` announces that the library was never found:
```
write(1, "-- Could NOT find FFmpeg (missing: FFMPEG_LIBRARIES FFMPEG_INCLUDE_DIRS AVCODEC_LIBRARIES AVCODEC_INCLUDE_DIRS AVFORMAT_LIBRARIES AVFORMAT_INCLUDE_DIRS AVUTIL_LIBRARIES AVUTIL_INCLUDE_DIRS SWSCALE_LIBRARIES SWSCALE_INCLUDE_DIRS) \n", 230) = 230
```

I think that sdrangel's `cmake/Modules/FindFFmpeg.cmake` is doing something wrong and needs fixing.

nemanjan00, 2024-08-03:
I had to disable `demoddatv` in AUR package, for it to be able to compile

https://aur.archlinux.org/packages/sdrangel-git

Quick solution being

```
		-DENABLE_CHANNELRX_DEMODDATV=OFF \
		-DENABLE_CHANNELTX_MODDATV=OFF
```

for cmake

dreirund, 2024-09-13:
> *This issue is going to be closed due to inactivity*

@github-actions, please do not close issues just because the maintainer is not active.

Rather, ask the maintainer @f4exb to look at the issue and decide about it's validity and resolvedness.

Regards!

dreirund, 2024-10-25:
> *This issue is going to be closed due to inactivity*

@github-actions, @f4exb: That does not solve any issues.

Please do not close issues just because the maintainers did not find time yet to react on them.

nemanjan00, 2024-12-05:
Once again, nope

dreirund, 2024-12-05:
> > This issue is going to be closed due to inactivity
> 
> *Once again, nope*

I wrote them an email to ask what the state of this is.

f4exb, 2024-12-06:
Firstly as a matter of courtesy for people watching this space please change the horrible bold title of this issue.

Then if this did not receive any attention it means that this issue is not impacting or interesting many people and you could probably solve it yoursef.

dreirund, 2024-12-06:
Ahoj,

> *Firstly as a matter of courtesy for people watching this space please change the horrible bold title of this issue.*

OK, I am going to remove the "* and warning `AutoUic: /var/cache/makepkg/build/sdrangel-git/src/sdrangel/plugins/channelrx/demodft8/ft8demodgui.ui: Warning: The name 'verticalLayout_2' (QVBoxLayout) is already in use, defaulting to 'verticalLayout_21'`*" part.

> *Then if this did not receive any attention it means that this issue is not impacting or interesting many people and you could probably solve it yoursef.*

It shouldn't be offending. When this issue is not relevant for you, why not you just close it as "not relevant for me", "no intention to fix", or so?  
Then everyone knows that it intentionally is not going to be done anything about it, and we would leave it as it is.  
Having a "timeout-bot" closing this issue does not give any information weather this issue is not going to be dealt with by the developers, or weather they just did not find time yet -- especially that you do it in spare time can mean that it could take years until you find the time (which I find totally legitimate), which a "timeout-bot" would just not allow for.

Regards!

dreirund, 2024-12-06:
Ahoj,

> *Firstly as a matter of courtesy for people watching this space please change the horrible bold title of this issue.*

OK, I am going to remove the "*; and warning `AutoUic: /var/cache/makepkg/build/sdrangel-git/src/sdrangel/plugins/channelrx/demodft8/ft8demodgui.ui: Warning: The name 'verticalLayout_2' (QVBoxLayout) is already in use, defaulting to 'verticalLayout_21'`*" part.  
Does this fit what you think of?

> *Then if this did not receive any attention it means that this issue is not impacting or interesting many people and you could probably solve it yoursef.*

It shouldn't be offending. I ask for forgiving if it was.

When this issue is not relevant for you, why not you just close it as "not relevant for me", "no intention to fix", or so?

Then everyone knows that it intentionally is not going to be done anything about it, and we(?) would leave it as it is.

Having a "timeout-bot" closing this issue does not give any information weather

* this issue is not going to be dealt with by the developers; or
* weather they just did not find time yet -- especially that you do it in spare time can mean that it could take years until you find the time (which I find totally legitimate), which a "timeout-bot" would just not allow for; or
* weather you just do not get any notification of it (I also already had it that someone with a git repository was just not getting information about issues and actually was very happy to be pointed to them),
* &hellip;

Regards!

calimeroteknik, 2025-01-16:
It will not, because a resolution timeline is dependent on new events, rather than wall time.

calimeroteknik, 2025-01-24:
Whoops! It is solved in the master branch indeed, I should have checked. Indeed certainly thanks to https://github.com/f4exb/sdrangel/commit/5103b270918e362907445c64e328032755bab225.
This issue can be closed!

closed 2025-01-24 by dreirund

dreirund, 2025-01-25:
It seems it has been solved silently.

I closed this issue.