f4exb/sdrangel #2207: Windows 11

HB9RYZ, 2024-07-11:
Hi All
SDRangel version sdrangel-7.21.4-win64.exe works only on **one** of threr Windows 11 PCs/Notebook.

I start SDRangel, but nothing will happen - no error message - just nothing.

Hard to analyze and to find out the issue?

Any idea?

73 de HB9RYZ
Wolfgang

srcejon, 2024-07-11:
Try installing the latest Visual C libraries from: https://aka.ms/vs/17/release/vc_redist.x64.exe

HB9RYZ, 2024-07-11:
Thanks "srcejon", you are the best. Now SDRangel works :-)
73 de HB9RYZ
Wolfgang

OP25Rookie, 2024-07-22:
I also had the same issue and this resolved it.  SDRAngel would not open.  Running it only showed the opening splash screen for a couple seconds. Then the splash screen disappears and the program never opened.  Running that .exe installer got it working now.  Thanks all.

closed 2024-09-07 by github-actions[bot]