f4exb/sdrangel #2209: Unable to open software for nooelec SDR V5 on iMac.

Kristiana-PG, 2024-07-12:
Hallo!
I am unable to open software for my nooelec SDR V5 on my iMac, I am using Sonoma 14.5.

I have tried using CubicSDR but it will not stay open, which software will work for me?

TY  :-) 

[image]

Axpelle, 2024-07-12:
Works fine for me on Sonoma MacMini using M2 chip. Make sure you download the right version for your setup.

Kristiana-PG, 2024-07-14:
Hi, which version did you use?

Axpelle, 2024-07-14:
I’m using the latest for M2 chip. 
https://github.com/f4exb/sdrangel/releases/download/v7.21.4/sdrangel-7.21.4-macarm64.dmg

Intel-based macs need the macx64 version.

———— Original message received from "Kristiana-PG" 
***@***.***> on 14/07/2024 17:03:48 ————

Subject Re: [f4exb/sdrangel] Unable to open software for nooelec SDR V5 
on iMac. (Issue #2209)

Kristiana-PG, 2024-07-14:
I have opened it, but there is just a grey screen and nothing else.

Axpelle, 2024-07-14:
At least the software opens. Is it a grey or black screen?

———— Original message received from "Kristiana-PG" 
***@***.***> on 14/07/2024 19:25:58 ————

Subject Re: [f4exb/sdrangel] Unable to open software for nooelec SDR V5 
on iMac. (Issue #2209)

Kristiana-PG, 2024-07-14:
Hi yessss! it's grey screen.

Axpelle, 2024-07-14:
When opened for the first time you will get an empty screen. You need to 
add a device manager first, then a demodulator for the type of signal 
you want to receive. Have a look at this video:

https://www.youtube.com/watch?v=zrhBcy8L-dA

———— Original message received from "Kristiana-PG" 
***@***.***> on 14/07/2024 19:52:49 ————

Subject Re: [f4exb/sdrangel] Unable to open software for nooelec SDR V5 
on iMac. (Issue #2209)

Kristiana-PG, 2024-07-14:
There is only a great screen with no headers or anything to click on?

Axpelle, 2024-07-15:
Can you upload a screenshot?

———— Original message received from "Kristiana-PG" 
***@***.***> on 14/07/2024 23:58:14 ————

Subject Re: [f4exb/sdrangel] Unable to open software for nooelec SDR V5 
on iMac. (Issue #2209)

Kristiana-PG, 2024-07-15:
[image]

Here we are..

Axpelle, 2024-07-16:
> Screenshot.2024-07-15.at.11.48.02.png (view on web)
> Here we are..

Definitely something wrong. There should be a row of small icons visible 
starting from the top left edge. Yet I can see from the bottom right 
edge that the programme is indeed open.

Only the developer can help in this case. You could raise a query on the 
forum.

Good luck

———— Original message received from "Kristiana-PG" 
***@***.***> on 15/07/2024 14:49:51 ————

Subject Re: [f4exb/sdrangel] Unable to open software for nooelec SDR V5 
on iMac. (Issue #2209)

Axpelle, 2024-07-17:
Have a look a the “Docs” section on the homepage. There it is mentioned 
“ If the toolbar is not visible, you can create a workspace by clicking 
on Workspaces then New”

Try it!

———— Original message received from "Axel Pellegrin" 
***@***.***> on 16/07/2024 07:33:15 ————

Subject Re[2]: [f4exb/sdrangel] Unable to open software for nooelec SDR 
V5 on iMac. (Issue #2209)

closed 2024-09-02 by github-actions[bot]