f4exb/sdrangel #2213: ATV Crashes SDRangel

JoeyTheGamer5670, 2024-07-19:
Help, whenever I try to use ATV, SDRangel crashes. Device is Infinix X6815C (Infinix Zero 5G 2023)

srcejon, 2024-07-19:
Can you add a few more details please? Mod or demod? What SDR? When exactly does it crash? What settings were you using? A screenshot just before would be useful.

JoeyTheGamer5670, 2024-07-22:
I have the video but it can't be sent to GitHub

JoeyTheGamer5670, 2024-07-22:
I'll send [Gdrive link](https://drive.google.com/file/d/1JQ-BgAOgGcrTV-_V8VdhGyZIMPunvpsn/view?usp=drivesdk)

JoeyTheGamer5670, 2024-08-06:
Hey, this is not something to ignore, please respond

srcejon, 2024-09-22:
Reproducible on Linux. 

- Use Test Source
- Press Start
- In ATV Demod, set L to 32.

2024-09-22 09:13:11.120 (D) TVScreenAnalog::resizeTVScreen: cols: 49, rows: -17
2024-09-22 09:13:11.120 (W) Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt.
You must not let any exception whatsoever propagate through Qt code.
If that is not possible, in Qt 5 you must at least reimplement
QCoreApplication::notify() and catch all exceptions there.

terminate called after throwing an instance of 'std::bad_array_new_length'
  what():  std::bad_array_new_length

Thread 50 "QThread" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffeaa00640 (LWP 2924)]
__pthread_kill_implementation (no_tid=0, signo=6, threadid=140737129743936) at ./nptl/pthread_kill.c:44
44	./nptl/pthread_kill.c: No such file or directory.
(gdb) info stack
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=140737129743936) at ./nptl/pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=140737129743936) at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=140737129743936, signo=signo@entry=6) at ./nptl/pthread_kill.c:89
#3  0x00007ffff5242476 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#4  0x00007ffff52287f3 in __GI_abort () at ./stdlib/abort.c:79
#5  0x00007ffff56a2b9e in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff56ae20c in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007ffff56ae277 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#8  0x00007ffff5a8ff91 in qt_assert(char const*, char const*, int) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#9  0x00007ffff5a9199e in  () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#10 0x00007ffff5294ac3 in start_thread (arg=<optimised out>) at ./nptl/pthread_create.c:442
#11 0x00007ffff5326850 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
(gdb) 

I guess TVScreenAnalog resizing rows to -17 is problematic

JoeyTheGamer5670, 2025-01-20:
I think it can't render anything, I have a Dimensity 1080 and 8 gigs, but not even any type of config can make the ATV run

JoeyTheGamer5670, 2025-01-26:
still not fixed

closed 2025-01-26 by JoeyTheGamer5670