f4exb/sdrangel #2220: DATV Module showing erratic movement

ramschulting, 2024-07-22:
I've been comfortably using SDRangel for quite a while and love the software. Am using it only for DATV reception of fellow HAMs but also for monitoring my own signal. There's however one issue I cant seem to solve. The decoding works just fine but the decoded video shows an erratic movement of e.g. my tickerline. I always thought it had to do with my back then not so powerfull GPU but last week I installed another NVIDIA card (GeForce GTX 1650) and with this card it still shows the same erratic behaviour.

I've upgraded the complete setup softwarewise, just in case, but nothing helps. Today I installed SDRangel on my Lenovo E570, this one has a Core I7  together with a NVIDIA GeForce GTX 950M, so pretty powerfull as well. But again I do have the same issue, tickerline erratic movement. I'm using a RTL-Play RSP-1A but also tried another SDR stick, both showing the same issue.

The issue is now also confirmed by fellow ham operators and seems to be concentrating on DVB-S2 only als DVB-S apparently doesn't seem to have the issue (or less).

It looks like some sort of a buffering issue, the same rithm is also vissible within the update frequency of the Constellation and the Counters as well.

By the way, when using the UDP stream sourced by the DATV module with VLC the issue is much less or even not visible.

I recorded an example and posted it to YouTube:
https://youtu.be/zgRUaVsGNVU

I hope this issue can be solved?
Kind regards,
Richard (PA1RAM)

f4exb, 2024-07-23:
The internal player has not the same capabilities as dedicated software like VLC. For best experience indeed it is recommended to use the UDP transport stream streaming.

ramschulting, 2024-07-23:
Thank you for your kind reply and making things clear. I will nolonger pursue on the issue and use VLC instead.

closed 2024-09-08 by github-actions[bot]