f4exb/sdrangel #2223: ADS-B Open Notification Dialog

Comm-on-call, 2024-07-26:
What is the syntax used to make use of the audio notifications? I am having trouble finding anything more than what is on the wiki. I would like to to use the language and command strings accurately.

srcejon, 2024-07-26:
All it does it that simple substitutions. What are you hoping for?

srcejon, 2024-07-26:
That isn't supported currently I'm afriad

Comm-on-call, 2024-07-26:
That's fine. What "syntax" is used under the "Reg Exp" column? I'm looking for "broad information" not trying to have my hand held necessarily. The wiki image give the impression there is the ability to filter to a certain extent.

srcejon, 2024-07-27:
Sorry, I misunderstood. I thought you were just referring to the substitutions. Reg Exps are Regular Expressions and should be broadly compatible with Perl syntax.

See: 

https://en.wikipedia.org/wiki/Regular_expression
https://regexr.com/
https://pcre.org/original/doc/html/pcrepattern.html

Comm-on-call, 2024-07-27:
Fantastic. This was what I looking for. I appreciate it.

closed 2024-08-28 by srcejon