f4exb/sdrangel #2224: no arm64 pkgs or support

N4CNR, 2024-07-30:
As arm is now a part of real world rhardware when do you plan to make arm64 pkgs ?

f4exb, 2024-07-30:
> As arm is now a part of real world rhardware when do you plan to make arm64 pkgs ?

Never unless you set up your own toolchain for the community. I have enough to do with new features and stability. The few artifacts found here are either the products of the CI./CD or the outcome of processes owned by their users.

> Also with building linux does not use /opt by default /opt is out dated unix dir.

This kind of bold statement is irrelevant. If you build from source and install in your system I strongly recommend not to do this as root so /usr/local is a no go. However if you compile from source in your own system you are the boss so you may install things anywhere you like.

alphafox02, 2024-08-26:
SDRangel is included in DragonOS Pi64 (aarch64) and as F4exb mentions, it's not terribly difficult to follow the excellent SDRAngel guide for Linux to compile and make a package yourself. It's better to compile with a 8GB Pi, but you can get away with 4GB with some fine tuning. I'm sure on other arm64 targets it'll compile just fine.

closed 2024-10-11 by github-actions[bot]