f4exb/sdrangel #2229: sdrangel crashes when loading a workstation

Degamisu, 2024-08-14:
I cannot load any of my workstations because it either stops at 25% or 50%. Mostly when loading things. When this gets stuck, around 3 seconds later it crashes.

srcejon, 2024-08-28:
You can run with an empty workspace, using --scratch command line option.

What are in your workspaces? Can you post the log file (Preferences > Logging) when it crashes?

Degamisu, 2024-08-28:
> You can run with an empty workspace, using --scratch command line option.
> 
> What are in your workspaces? Can you post the log file (Preferences > Logging) when it crashes?

Sure! I've tried every workspace (Factory as well) and the issue persists. For example, the ISS packet factory workspace does the same thing.

I just tried it now - it seems to be working? It isn't crashing anymore. I'll try and make it crash again. Could be a memory issue.

closed 2024-10-13 by github-actions[bot]