f4exb/sdrangel #2231: LibSigMF compile fails only for SDRangel

srs4511351, 2024-08-14:
Using a Raspberry Pi 5
The LibSigMF installation fails when I compile it with the SDRangel plugin procedure on Ubuntu 24.04 LTS
The same procedure works with Raspberry Pi OS. I suspect that the newer packages used in Ubuntu 24.04 are the problem.
```
cd ~/SDRangel
git clone https://github.com/f4exb/libsigmf.git
cd libsigmf
git checkout "new-namespaces"
mkdir build; cd build
cmake -Wno-dev -DCMAKE_INSTALL_PREFIX=/opt/install/libsigmf .. 
make -j $(nproc) install
```
The failure occurs during the compiling of flatbuffers.
I can successfully compile flatbuffers with a procedure similar to the one on https://github.com/google/flatbuffers. I don't know what is different in the SDRangel procedure.
```
cd /home/pi/SDRangel/libsigmf/external
git clone https://github.com/google/flatbuffers.git
cd flatbuffers
cmake -DCMAKE_INSTALL_PREFIX=/home/pi/SDRangel/libsigmf/build/flatbuffers-build .
make -j4
make install
```
I have attached the terminal output for the SDRangel LibSigMF compile.

I also installed the libsigmf/noble,now 1.0.2-2 arm64 package which installed the flatbuffers packages, but SDRangel did not find it even after I removed -DLIBSIGMF_DIR=/opt/install/libsigmf  from the SDRangel cmake statement.

[LibSigMF compile 8-14-2024.txt](https://github.com/user-attachments/files/16618754/LibSigMF.compile.8-14-2024.txt)

srcejon, 2024-08-28:
My guess is external/flatbuffers link in f4exb/libsigmf points to an old version of flatbuffers (Seems to be the tag [d8210d5](https://github.com/google/flatbuffers/tree/d8210d5a83faa345046648e520d82b54ea724e35)) that doesn't compile with the latest version of gcc. When you build it yourself, you get the latest version.

So the link to flatbuffers probably needs updating.

srs4511351, 2024-08-28:
tag d8210d5 is indeed quite old.

Will you update the link or tell me how to do it on my system?
libsigmf/external on my system does not have the links, just empty folders.

----Steve

closed 2024-10-13 by github-actions[bot]

srs4511351, 2024-11-10:
I see that there is a fix for Fix for flatbuffers v2: https://github.com/f4exb/libsigmf/commit/7d964fdf3fc50075e66a705633eaf58247043c59 on [on the original branch](https://github.com/deepsig/libsigmf)