f4exb/sdrangel #2242: Radiosonde Feature Plugin loses connection with SondeHub

GabrielSampol, 2024-09-01:
I am testing the Radiosonde Feature Plugin in the latest version of SDRangel (Qt 5.15.2 65 bits) on Windows 11 and using an RTL V4 dongle.

It works nicely, except that it seems to lose the connection with SondeHub after running for a while.

The steps to recreate the issue es:
- Radiosonde decoding starts and frames are beeing received and shown by the plugin and track on the maps.
- My callsign appears OK in SondeHub as receiver of the radiosonde
- After a short while my callsign dissapears in SondeHub as if I was not sending data, but frames are being received and processed by the plugin.
- If I close SDRangel and start it again, SondeHub list me again as receiver, but also again stops doing it after a short time.

Is this a known issue?  If you want me to make some additional test just let me know.

GabrielSampol, 2024-09-14:
No one else is having this issue? Strange I haven't get any answer....

GabrielSampol, 2025-01-25:
I haven't used the program since September. Where is the "Feed button" so that I can configure the upload to Sondehub?  I can't find it.

GabrielSampol, 2025-01-25:
Trying it. Will let you know

GabrielSampol, 2025-01-25:
I confirm that the problem persists. Exactly the same behavior.

srcejon, 2025-01-25:
Strange, it works for me.

Do you know the serial number of the sonde you were tracking, so I can look in the SondeHub database? (Or  roughly what location it is on the map?) Thanks

GabrielSampol, 2025-01-26:
I will try to make another test today at 12 UTC and give you more detailed information.

What I notice is that in Sondehub for my callsign it doesn't show neither the signal strenght nor the frequeny in "Received via", while this data is shown for other callsigns. I don't know if this could have some relationship with the issue.

srcejon, 2025-01-26:
We don't send signal strength - but that wasn't a problem for me when I tried it last.

GabrielSampol, 2025-01-26:
Today it has been working for a long period but at about 13:08 (CET) it has stoped again.

I see in the log file that there are error about Pressure being below cero.

I attach the log file and also a short video where you can see the situation.

[sdrangel.log](https://github.com/user-attachments/files/18550030/sdrangel.log)
https://github.com/user-attachments/assets/db3daace-5fcc-4957-b47e-b6f08308335c

srcejon, 2025-01-26:
> I see in the log file that there are error about Pressure being below cero.

Thanks. Looks like a bug, as that's not a valid value.

Unfortunately, the Radiosondes near me don't actually transmit pressure measurements, so I can't test that easily. Is there any chance you can make an IQ recording at some point so I can try to fix it? You just need to Add Channel > File Sink then set a file name (E.g. radiosonde.wav) and press record. A couple of minutes should be enough.

GabrielSampol, 2025-01-26:
I will be pleased to make the recording, but where is the "Add channel" option?  I can't find it.

srcejon, 2025-01-26:
In the device window, to the left of where it says RTL SDR[0]. Same button you pressed to add the Radiosonde Demodulator.

GabrielSampol, 2025-01-26:
OK, thanks. Due to the limited lauch time of the sondes I will not be able to make the recording until tuesday noon. I will let you know when done.

GabrielSampol, 2025-01-28:
Github does now allow to attach wav files, so I have put everyhing in a compressed file that you can download from http://dxmaps.com/temp/Files_SDRangel.zip

It contains the log file, plus two wav files and two csv files with the received data.

File names starting with "OK" are when Sondehub accepts my data. Files names starting with "BAD" are when Sondehub rejects my data because of the negative values of pressure.

I hope they help you and that you fix the bug.  I plan monitoring 24/7 and l like SDRangel very much, so I would like to use if you can  fix it.

If you need some further tests, just let me know. 

Looking forward to your news. Thanks.

closed 2025-02-09 by f4exb

GabrielSampol, 2025-02-09:
Is the fix part of version 7.22.6?

Because, if so, it has not been fixed.

[image]

 fixed.  T

srcejon, 2025-02-09:
> Is the fix part of version 7.22.6?

No. Will be in the next release or you can try a test build: https://github.com/f4exb/sdrangel/actions/runs/13115247237/artifacts/2527605398

PR #2394 (2025-02-09): Radiosonde: Fix pressure calculation. 
Fix pressure calculation in Radiosonde demod. Signed value was erroneously extracted as unsigned. Fixes #2242