f4exb/sdrangel #2246: 3D map is blank on Raspberry Pi 5 and Raspberry Pi OS

srs4511351, 2024-09-06:
On a Raspberry Pi OS Bookworm system, 3D map works on a Raspberry Pi 4.
With the same image, booted on a Raspberry Pi 5 the 3D map is blank.

On a Raspberry Pi OS built by Raspberry Pi for a RPI 5, and with SDRangel installed on the RPI 5, the 3D map is blank.
With Ubuntu 24.04, on a Raspberry Pi 5, the 3D map works.

This is strikingly similar to a problem with GridTracker. On the RPI 5, it only works on OSes that are not Raspberry Pi OS or DietPI.

Does SDRangel 3D maps use NWJS? Here is some insight into the GridTracker issue.
https://groups.io/g/GridTrackerApp/topic/instructions_to_build_from/103035128
"GT itself is a NWJS application, so its essentially a web app that is mostly html and JavaScript. As such there isn't much to compile and what segfaults is the nwjs execution environment."

I attached the log file from running SDRangel with maps open, then I quit. I have the 3D models downloaded.
[sdrangel.log](https://github.com/user-attachments/files/16902081/sdrangel.log)

srcejon, 2024-09-06:
The 3D Map uses Cesium. 

Does this work for you? https://sandcastle.cesium.com/

srs4511351, 2024-09-06:
Yes, it does work in Firefox on my Raspberry Pi.

My Windows computer does not work in Google Chrome and Firefox. It suggests that I don't have WebGL

srcejon, 2024-09-06:
> Yes, it does work in Firefox on my Raspberry Pi.

Which one? 5 with Pi OS?

srs4511351, 2024-09-06:
It worked on both the RPI 5 with Dietpi OS and Raspberry Pi OS on the RPI 5. 3D maps are also blank in Dietpi.

srs4511351, 2024-10-17:
Sorry, my previous reply was confusing.
3D maps work on RPI 4.
3D maps do not work and are blank on RPI 5 Raspberry Pi OS, and Dietpi
They do work on Ubuntu.

srs4511351, 2024-11-27:
Is there any progress getting 3D maps working on the Raspberry Pi 5 running Raspberry Pi OS?

srcejon, 2024-11-27:
Nope - I don't have one to try. From your description, sounds like it's probably not an SDRangel specific problem anyway.

srs4511351, 2024-11-27:
Thanks for the update.
It does work on https://sandcastle.cesium.com/

srcejon, 2025-01-08:
The version of cesium we are using is a couple of years old, so I'll try updating it

srs4511351, 2025-02-18:
Please update version of cesium.
The issue is expiring...

Thanks, Steve

closed 2025-02-18 by srs4511351