f4exb/sdrangel #2248: ATV modulator no longer available in Windows build

pa3byv, 2024-09-08:
Hi, we are still using v6.16 in Windows 10 because it has an ATV modulator. It seems to have disappeared from Windows builds in V7, probably from V6.20 onwards. But I see it is still maintained and available in the Linux builds. Is there a special reason for this and if not could you please include it in the V7 Windows build again? 

Many thanks,
Poll

srcejon, 2024-09-09:
From the github action log, I see:

 Not building modatv (ENABLE_CHANNELTX_MODATV=ON OpenCV_FOUND=0)

I think this is probably due to the version of OpenCV in sdrangelwindowslibraries not working with the latest version of Visual C. Updating to 4.10 seems to work.

pa3byv, 2024-09-09:
Thanks for looking into it. Hopefully the ATV module will reappear in the next build then.

0x0V3RF10W, 2024-09-15:
Hello! Any plans to return modatv to windows builds?

srcejon, 2024-09-15:
@f4exb - unfortunately need a small patch to the opencv cmakefiles: https://github.com/f4exb/sdrangel-windows-libraries/pull/26

srcejon, 2024-09-15:
> Hello! Any plans to return modatv to windows builds?

Hopefully should be in the next release.

srcejon, 2024-10-11:
Back in 7.22.1

closed 2024-10-11 by srcejon

pa3byv, 2024-10-11:
Thanks very much!