f4exb/sdrangel #2249: GPSD support

bddvlpr, 2024-09-09:
Since my writing as of #1524, I've been thinking that maybe it'd be more helpful to support GPSD instead of just the serial connection. This would allow users to get position data from any source without SDRangel having to include any logic (besides reading a local net socket).

[More information about GPSD can be found here.](https://gpsd.gitlab.io/gpsd/)

What are your thoughts on this?

srcejon, 2024-09-10:
SDRangel uses Qt Positioning - so we support what that supports. 

On Linux, Qt Positioning supports Geo Clue 2 as well as NMEA via serial port. Different sources of position are implemented in plugins, so it would be possible to write a Qt Positioning plugin that supports GPSD. It could be based on other code here: https://github.com/qt/qtpositioning/tree/dev/src/plugins/position (or the NMEA plugin couldn't be modified to add TCP support).

Not something I'm personally interested in doing, but if you were to try, it could be added to the Qt source code (so all Qt apps would support it).

bddvlpr, 2024-09-10:
Apparently a few abandoned implementations exist. Running geoclue2 currently it _should_ pick up nmea sources like gpsd up automatically but I'm failing to do so. Nonetheless closing this issue as geoclue2 should be able to work with gpsd.

closed 2024-09-10 by bddvlpr