f4exb/sdrangel #2256: SDRAngel crashes with LimeSuite.dll accesing MSVCP140.dll on Windows

psygate, 2024-09-19:
Hi, I'm not sure where to put this but maybe someone else encounters the same problem.

SDRAngel 7.22.0 crashes for me on startup with a segmentation fault (0xc0000005), the dump analysis shows that the LimeSuite.dll is accessing MSVCP140.dll and fails when trying to read from a nullptr.

To fix this, I installed LimeSuite.dll from the official PothosSDR and removed the LimeSuite.dll from the SDRAngel folder. SDRAngel now starts again without issues.

srcejon, 2024-09-22:
Same issue as #2207 I think.

psygate, 2024-09-22:
> Same issue as #2207 I think.

I'm not sure it's exactly the same. I have the newest MS VC++ already installed, downloading the linked version only allows me to repair or uninstall. Do you build the dlls with the sdrangel binary together every time, or are they just packaged?

I'm happy to help here where I can, I love this application 😄

srcejon, 2024-09-23:
> I'm not sure it's exactly the same. I have the newest MS VC++ already installed, downloading the linked version only allows me to repair or uninstall.

See last comment in - https://github.com/f4exb/sdrangel/issues/2244  - although didn't try it myself.

> Do you build the dlls with the sdrangel binary together every time, or are they just packaged?

For LimeSuite, I think they are built everytime. We should probably update external/CMakeLists.txt to pick up the latest version though.

psygate, 2024-09-23:
I uninstalled the VC Runtime and reinstalled it, now SDRAngel is working and starting up normally again. Very strange, since Windows seems to think the new VC Runtime and the old one are the same and don't need a reinstall at all...

I also read the other thread, I have PothosSDR and LimeSuite installed, and a LimeSDR that has been connected to this PC. Really seems to be the runtime the LimeSuite was built against.

closed 2024-11-08 by github-actions[bot]