f4exb/sdrangel #2258: MacOS: Missing devices

ogermer, 2024-09-25:
I installed SDRAngel on macOS Sonoma 14.6.1
When adding the Rx / sampling device I see very few devices (see attached screenshot)
Popular devices like RTL-SDR or HackRF are missing.
I can use both on SDR++ and have the hackrf tools installed via brew.
I am using the [v7.22.0](https://github.com/f4exb/sdrangel/tree/v7.22.0) release.
[image]

srcejon, 2024-09-25:
Are the devices open in another application at the same time?
Is this an intel or ARM mac? For the latter, you might want to try 7.21 ARM build.
Can you post the Log file? (Preferences > Logging)

ogermer, 2024-09-25:
Hi - thanks for the quick response!
No - the device was not open with another app.
Yes - it is an M1 (ARM) Mac. I will try the (older) [v7.21.4](https://github.com/f4exb/sdrangel/tree/v7.21.4) which has an ARM version.

The log follows below:

`
2024-09-25 12:23:08.401 (I) SDRangel 7.22.0 Qt 5.15.2 64b x86_64 macOS 10.16 DSP Rx:24b Tx:16b PID 95852
2024-09-25 12:23:09.465 (W) MainCore::positionUpdateTimeout: GPS signal lost
2024-09-25 12:23:14.190 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.aaroniartsasource
2024-09-25 12:23:14.190 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.airspy
2024-09-25 12:23:14.190 (D) AirspyPlugin::enumOriginDevices: enumerated 0 Airspy devices AIRSPY_ERROR_NOT_FOUND
2024-09-25 12:23:14.191 (D) AirspyPlugin::enumOriginDevices: airspy_exit: AIRSPY_SUCCESS
2024-09-25 12:23:14.191 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.airspyhf
2024-09-25 12:23:14.191 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.audioinput
2024-09-25 12:23:14.191 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.bladerf1input
2024-09-25 12:23:14.192 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.bladerf2input
2024-09-25 12:23:14.193 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.fileinput
2024-09-25 12:23:14.193 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.fcdpro
2024-09-25 12:23:14.193 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.fcdproplus
2024-09-25 12:23:14.193 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.hackrf
2024-09-25 12:23:14.193 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.kiwisdrsource
2024-09-25 12:23:14.194 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.limesdr
2024-09-25 12:23:14.194 (D) DeviceLimeSDR::enumOriginDevices: Could not find any LimeSDR device
2024-09-25 12:23:14.194 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.localinput
2024-09-25 12:23:14.194 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.plutosdr
2024-09-25 12:23:14.201 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.rtlsdr
2024-09-25 12:23:14.201 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.remotetcpinput
2024-09-25 12:23:14.201 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.remoteinput
2024-09-25 12:23:14.202 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.testsource
2024-09-25 12:23:19.166 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.aaroniartsasource
2024-09-25 12:23:19.166 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.airspy
2024-09-25 12:23:19.166 (D) AirspyPlugin::enumOriginDevices: enumerated 0 Airspy devices AIRSPY_ERROR_NOT_FOUND
2024-09-25 12:23:19.166 (D) AirspyPlugin::enumOriginDevices: airspy_exit: AIRSPY_SUCCESS
2024-09-25 12:23:19.166 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.airspyhf
2024-09-25 12:23:19.167 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.audioinput
2024-09-25 12:23:19.167 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.bladerf1input
2024-09-25 12:23:19.168 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.bladerf2input
2024-09-25 12:23:19.168 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.fileinput
2024-09-25 12:23:19.168 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.fcdpro
2024-09-25 12:23:19.169 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.fcdproplus
2024-09-25 12:23:19.169 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.hackrf
2024-09-25 12:23:19.169 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.kiwisdrsource
2024-09-25 12:23:19.169 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.limesdr
2024-09-25 12:23:19.169 (D) DeviceLimeSDR::enumOriginDevices: Could not find any LimeSDR device
2024-09-25 12:23:19.169 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.localinput
2024-09-25 12:23:19.169 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.plutosdr
2024-09-25 12:23:19.176 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.rtlsdr
2024-09-25 12:23:19.176 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.remotetcpinput
2024-09-25 12:23:19.176 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.remoteinput
2024-09-25 12:23:19.176 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.testsource
2024-09-25 13:43:22.159 (D) MainWindow::closeEvent
2024-09-25 13:43:22.159 (D) MainWindow::saveConfiguration: configuration [default | no name] 1 workspaces
2024-09-25 13:43:22.160 (D) AudioDeviceManager::serialize
2024-09-25 13:43:22.351 (I) SDRangel 7.22.0 Qt 5.15.2 64b x86_64 macOS 10.16 DSP Rx:24b Tx:16b PID 95852
2024-09-25 13:43:22.362 (D) MainWindow::~MainWindow
2024-09-25 13:43:22.362 (D) HttpListener: closed
2024-09-25 13:43:22.362 (D) HttpConnectionHandlerPool (0x600003dc2450): destroyed
2024-09-25 13:43:22.362 (D) HttpListener: destroyed
2024-09-25 13:43:22.362 (I) WebAPIServer::stop: stopped web API server at http://0.0.0.0:8091
2024-09-25 13:43:22.363 (D) Workspace::~Workspace
2024-09-25 13:43:22.363 (D) Workspace::~Workspace: about to delete MDI
2024-09-25 13:43:22.363 (D) Workspace::~Workspace: end
2024-09-25 13:43:22.363 (D) MainWindow::~MainWindow: end
`

ogermer, 2024-09-25:
I tried sdrangel-7.21.4-macarm64.dmg bit with the same result.

konart, 2024-10-01:
Just FYI: I have the same problem. BUT while SDRAngel does not list RTL-SDR when trying to add Rx device - RTL-SDR is actually works fine if you load a configuration from Configurations window.

srcejon, 2024-10-01:
Can you post a full screen shot please?

Note that the RTL SDR will not be listed if it's already open somewhere (possibly off-screen or in a different workspace)

closed 2024-11-16 by github-actions[bot]