f4exb/sdrangel #2262: Program crashes (Core Dump) when trying to add a Frequency Scanner channel

jtaylor2, 2024-09-30:
Program crashes when trying to add a Frequency Scanner channel to the opened program.
Or if the last closed program was from before the bug and had a Frequency Scanner channel it crashes on startup until the saved configuration is deleted.

Self compiled master on openSUSE Leap 15.6
Qt - 5.15.12  gcc11-c++ - 11.3.0

Git bisect says bad commit is [d2e89b733721432820d60442fe18d0e32cccd7ea] Fix PR comments

Bisecting: 0 revisions left to test after this (roughly 0 steps)
[d2e89b733721432820d60442fe18d0e32cccd7ea] Fix PR comments

gdb 
Thread 73 "QThread" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff73fff6c0 (LWP 27561)]
0x00007fffb0ca9968 in FreqScanner::calcScannerSampleRate(int, int, int&, int&, int&) ()
   from /home/jim/download/radio/sdrangel/sdrangel/build/lib/plugins/libfreqscanner.so

jtaylor2, 2024-10-03:
Backing out the changes in this diff seems to correct the problem, but my 1970's K&R C experience doesn't give me the ability to fix the intent of the original change.
```
diff --git a/plugins/channelrx/freqscanner/freqscanner.cpp b/plugins/channelrx/freqscanner/freqscanner.cpp
index 48708d72a..e5d62b59f 100644
--- a/plugins/channelrx/freqscanner/freqscanner.cpp
+++ b/plugins/channelrx/freqscanner/freqscanner.cpp
@@ -153,7 +153,7 @@ void FreqScanner::start()
 
     qDebug("FreqScanner::start");
     m_thread = new QThread();
-    m_basebandSink = new FreqScannerBaseband();
+    m_basebandSink = new FreqScannerBaseband(this);
     m_basebandSink->setFifoLabel(QString("%1 [%2:%3]")
         .arg(m_channelId)
         .arg(m_deviceAPI->getDeviceSetIndex())
diff --git a/plugins/channelrx/freqscanner/freqscannerbaseband.cpp b/plugins/channelrx/freqscanner/freqscannerbaseband.cpp
index 4fd211e2b..30357b2a4 100644
--- a/plugins/channelrx/freqscanner/freqscannerbaseband.cpp
+++ b/plugins/channelrx/freqscanner/freqscannerbaseband.cpp
@@ -27,7 +27,9 @@
 
 MESSAGE_CLASS_DEFINITION(FreqScannerBaseband::MsgConfigureFreqScannerBaseband, Message)
 
-FreqScannerBaseband::FreqScannerBaseband() :
+FreqScannerBaseband::FreqScannerBaseband(FreqScanner *freqScanner) :
+    m_freqScanner(freqScanner),
+    m_sink(),
     m_messageQueueToGUI(nullptr)
 {
     qDebug("FreqScannerBaseband::FreqScannerBaseband");
diff --git a/plugins/channelrx/freqscanner/freqscannerbaseband.h b/plugins/channelrx/freqscanner/freqscannerbaseband.h
index 5a832ff41..2fcdc2d7a 100644
--- a/plugins/channelrx/freqscanner/freqscannerbaseband.h
+++ b/plugins/channelrx/freqscanner/freqscannerbaseband.h
@@ -63,7 +63,7 @@ public:
         { }
     };
 
-    FreqScannerBaseband();
+    FreqScannerBaseband(FreqScanner *freqScanner);
     ~FreqScannerBaseband();
     void reset();
     void feed(const SampleVector::const_iterator& begin, const SampleVector::const_iterator& end);
```

f4exb, 2024-10-03:
Probably this PR deserves a second look

f4exb, 2024-10-03:
After double checking only the Frequency Scanner has this bug. I think the cleanup went a bit too far there where it is justified for other plugins.

commit 9d1160e7 (2024-10-04): Frequency Scanner: initialize pointer to FrequencyScanner. Fixes #2262
closed 2024-10-04 by f4exb