f4exb/sdrangel #2268: Significant performance degradation in FFTW initialization on ARM processor under Ubuntu 22.04

zhangxianhong, 2024-10-07:
I hope this message finds you well. I am writing to report a performance issue I have encountered while using SDRAngel on an ARM processor running Ubuntu 22.04. Specifically, I have noticed that the initialization of FFTW takes approximately 5 minutes, which is a significant departure from the instant startup time observed on x86 CPUs.

To provide more context, I have attached the complete startup log, which includes the time taken by FFTW for initialization. Additionally, I have included a log detailing the duration of the FFTW initialization process and some basic system information about my setup.

FFTWF log(
[Uploading run.log…]()

:
```text
2024-10-07 20:20:33.994 (D) FFTEngine::create: using FFTW engine
2024-10-07 20:20:33.994 (D) FFTWEngine::configure: no FFTW wisdom file
2024-10-07 20:20:54.072 (D) FFT: creating FFTW plan (n=128,forward) took 20077 ms
2024-10-07 20:20:54.072 (D) FFTEngine::create: using FFTW engine
2024-10-07 20:20:54.072 (D) FFTWEngine::configure: no FFTW wisdom file
2024-10-07 20:21:37.669 (D) FFT: creating FFTW plan (n=256,forward) took 43597 ms
2024-10-07 20:21:37.669 (D) FFTEngine::create: using FFTW engine
2024-10-07 20:21:37.670 (D) FFTWEngine::configure: no FFTW wisdom file
2024-10-07 20:22:55.335 (D) FFT: creating FFTW plan (n=512,forward) took 77665 ms
2024-10-07 20:22:55.335 (D) FFTEngine::create: using FFTW engine
2024-10-07 20:22:55.335 (D) FFTWEngine::configure: no FFTW wisdom file
2024-10-07 20:24:56.846 (D) FFT: creating FFTW plan (n=1024,forward) took 121510 ms

```

My system configuration is as follows:
- Processor:   
```sh
cat /proc/cpuinfo
processor	: 7
BogoMIPS	: 48.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x4
CPU part	: 0xd0b
CPU revision	: 0
```
- Operating System: 
```sh
uname -a
Linux FILY-RK3588 5.10.160 #1 SMP Sun Aug 25 21:04:24 CST 2024 aarch64 aarch64 aarch64 GNU/Linux
```
- SDRAngel Version:  7.22.0

Steps to reproduce:
1. Start SDRAngel on the specified system configuration.
2. Observe the time taken for FFTW initialization.

Expected behavior:
The FFTW initialization should complete in a matter of seconds, similar to the performance seen on x86 CPUs.

Actual behavior:
The FFTW initialization process takes around 5 minutes to complete.

Logs and additional information can be found in the attachments. I would appreciate any guidance or suggestions on how to resolve this performance issue. Thank you for your attention to this matter, and I look forward to your response.

srcejon, 2024-10-07:
Try generating a FFT Wisdom file, using Preferences > FFT Wisdom

f4exb, 2024-10-07:
I suppose RK3588 is a SBC similar to RPi. On these machines it is mandatory to use a FFT Wisdom file.

zhangxianhong, 2024-10-08:
> Try generating a FFT Wisdom file, using Preferences > FFT Wisdom
Thanks for the suggestion! I've tested the solution and it works perfectly. Appreciate the help!

zhangxianhong, 2024-10-08:
The issue has been resolved by following the method suggested in [#2397031014]. Closing this issue now. Thanks, everyone!

closed 2024-10-08 by zhangxianhong