f4exb/sdrangel #2272: Feature request: TDoA / multilateration support

rkarikari, 2024-10-09:
Is it possible now (or in the near future) to geolocate a transmitter with SDRAngel?

It runs on gps enabled devices ( phones, tablets etc) and can use these for TDoA calculation.

multilateration should also be possible for modeS only planes to be plotted on the map.

This would be a great feature addition .

srcejon, 2024-11-12:
Possibly, but you need multiple antennas in different locations.

The ADS-B demod already has the ability to feed packets to a server over IP - so we could possibly add a Feature that runs a server that takes those packets from multiple clients and then performs the TDoA calcs.

But not many users probably have 4 SDRs in different locations...

rkarikari, 2024-11-12:
> Possibly, but you need multiple antennas in different locations.
...
> But not many users probably have 4 SDRs in different locations...

Yes, you are right. 
 ... but I imagine we could choose the extra ( 3 or more ) from publicly available servers (as kiwisdr does) .  
Also the "sampling device" could have the option of additional "Addresses" for the other servers for TDoA?

[image]

srcejon, 2024-11-12:
You don't really want to do it by streaming IQ data at 2Mb/s. It can just about be done with RemoteTCP's IQ compression and squelch - but far more sensible just to send decoded packets.

closed 2025-01-02 by github-actions[bot]