f4exb/sdrangel #2278: use of a USB Midi Device as Hardware Controller

marqzB, 2024-10-12:
Is there a possibility to connect a Midi hardware controller like the mackie standard compatible controllers like the Behringer X Touch Series as a interface for the SDR Software?
It would be nice if you could map any Midi controller to the Buttons of your software like in any DAW like Ableton etc. so you could use a cheap Controller and make the software more Hardware alike.

f4exb, 2024-10-12:
You have this plugin: https://github.com/f4exb/sdrangel/tree/master/plugins/feature/jogdialcontroller If you are able to map your controller to keyboard key sequences then it may work.

marqzB, 2024-10-13:
i Doubt this will work the midi device is no keyboard / HID Device, but maybe there is a library that does this.

(Another Idea, can I use an VST to insert a Audio Filter in the NF path?)

closed 2024-11-28 by github-actions[bot]

srcejon, 2026-09-27:
> (Another Idea, can I use an VST to insert a Audio Filter in the NF path?)

Good idea - now implemented in the Denoiser feature: https://github.com/f4exb/sdrangel/pull/2958