f4exb/sdrangel #2279: How to connect HackRF One via RemoteTCPInput in SDRangel?

zhangxianhong, 2024-10-13:
Hello,
I am a user of SDRangel and I have both a HackRF One device and an RTL-SDR device. Currently, I have successfully connected my RTL-SDR device to SDRangel using the RemoteTCPInput feature by converting the RTL-SDR signal into TCP packets with the rtl_tcp command. However, I am facing difficulties in connecting my HackRF One device in the same manner.
Could someone please guide me on how to achieve this? I would like to know if there are any specific steps or additional configurations required to use HackRF One with RemoteTCPInput in SDRangel.
Thank you for your help in advance!
Best regards

srcejon, 2024-10-13:
Open tbe HackRF device, then add a Remote TCP Sink channel.

https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/remotetcpsink/readme.md

zhangxianhong, 2024-10-13:
> Run SDRangel on the remote device, open tbe HackRF device, then add a Remote TCP Sink channel.
> 
> https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/remotetcpsink/readme.md

Great, that's an effective method. Is it possible to achieve this without using the "Remote TCP Sink channel" (running SDRAngel on HackRF Device)? Just like how 'rtl_tcp' can simply implement RemoteTCPInput with just one command line? I would prefer that much more.

srcejon, 2024-10-13:
You can run SDRangel in server mode (so it has no GUI). E.g: something like:

    sdrangel --remote-tcp --remote-tcp-hwtype HackRF

To see the list of hwtypes, use --list-devices

f4exb, 2024-10-13:
> running SDRAngel on HackRF Device

no... you need a computer or SBC (e.g. Raspberry-Pi) with the HackRF connected to it.

zhangxianhong, 2024-10-14:
> You can run SDRangel in server mode (so it has no GUI). E.g: something like:
> 
> ```
> sdrangel --remote-tcp --remote-tcp-hwtype HackRF
> ```
> 
> To see the list of hwtypes, use --list-devices

I solved my problem with the method you provided, thank you.

closed 2024-10-14 by zhangxianhong