f4exb/sdrangel #2282: Spectrum measurements inconsistent in Broadcast FM Demod plugin

rapgenic, 2024-10-15:
Hi,

I am using my PlutoSDR to do some qualitative measurements on my FM transmitter. I am having it generate a 10KHz modulated FM signal, receiving it with my SDR and demodulating it with the SDRAngel Broadcast FM Demod plugin.

I am interested in a few measurements provided by the spectrum analyser, which are, the peaks and the SNR.

The SNR measurement produces totally nonsense results, but does not provide any marker inside the spectrum, so I'm not reporting it yet because it is not easy to show you.

However when visualizing peaks it is clear that there is some kind of problem. In the following image I have reported the peak measurement for both the RF and baseband spectra, clearly showing that in the BB spectrum all peaks seem to be detected on the same fundamental tone, despite all the peaks being clearly "visible".

[image]

I do not think I am doing something wrong, I have tried multiple spectrum parameters (i.e. averaging and FFT resolution) but I suspect they are not relevant

srcejon, 2024-10-15:
You have an 8k FFT - but the image is only 700 pixels wide. I'd guess if you zoom in you'll see multiple peaks.

rapgenic, 2024-10-15:
[image]
[image]

Here are the zoomed in version of the fundamental tone, at 8k and 32k ffts... I believe those are single peaks, is there some other way to check whether I'm wrong?

srcejon, 2024-10-15:
Try saving the spectrum data to .csv then look at the actual values.

In the last image, look at peak 15 - it may be it is fractionally higher than the point to it's left, but not a real peak. If that is the case, then perhaps we need some parameter that specifies a minimum height, in order to filter out things like that.

rapgenic, 2024-10-15:
Thanks for the suggestion!
I have plotted the result for the 32k FFT around the 10Khz tone. The procedure for acquiring this data was:
1. Run acquisition
2. Adjust all parameters (i.e. zoom, fft size, etc)
3. Stop the acquisition
4. Take a screenshot of the BB spectrum
5. Save the CSV spectrum content and plot the 10KHz peak

[image]
[image]

Here I can see a few interesting things:
* All values are duplicated, Maybe the peak finder algorithm considers a equal value as non decreasing, thus a possible peak? Also looking at the CSV ALL values seem to be duplicated, not just ones around the peaks, is this something to be expected as normal? 
* The peaks frequencies seem different from the CSV data, it seems that it's taking the middle point between two samples as the peak frequency (for example, look at peak 1, SDRangel reports -20.1 dB at 9.997Khz, whereas in the CSV we can see that we have -20.1 dB at 9994.39 and 10002 Hz, whose average is 9998 Hz). This is somewhat curious as well, but not inherently wrong I guess

srcejon, 2024-10-15:
> Maybe the peak finder algorithm considers a equal value as non decreasing, thus a possible peak

Yep, it does, in GLSpectrumView::peakWidth. If you are compiling SDRangel yourself, you can try changing:

    while ((left > maxLeft) && (spectrum[left] < prevLeft) && (right < maxRight) && (spectrum[right] < prevRight))

To:

    while ((left > maxLeft) && (spectrum[left] <= prevLeft) && (right < maxRight) && (spectrum[right] <= prevRight))

> Also looking at the CSV ALL values seem to be duplicated, not just ones around the peaks, is this something to be expected as normal? 

It looks like that's because it's a single-side-band spectrum. 

> The peaks frequencies seem different from the CSV data, it seems that it's taking the middle point between two samples as the peak frequency

Probably a difference in labeling: i.e. one is bin start frequency, whereas the other is the bin centre frequency.

rapgenic, 2024-10-15:
Hi, I finally managed to test a build with your suggested change and it actually works!

[image]

SNR measurement is working great as well!

[image]

I have seen that the measurement is performed only on the current span of the spectrum. Do you know if there is an easy way to set the span/center frequency without necessarily using the scroll wheel?

srcejon, 2024-10-15:
> Do you know if there is an easy way to set the span/center frequency without necessarily using the scroll wheel?

I don't think there is currently.

rapgenic, 2024-10-15:
I understand. 

Thank you very much for your support! 

If It's ok I'd like to open a PR with this fix, unless there are any side effects that I can't think of.

srcejon, 2024-10-18:
> If It's ok I'd like to open a PR with this fix, unless there are any side effects that I can't think of.

It should be OK - the alternative is that SSBs are handled separately.

PR #2299 (2024-10-31): Fix spectrum peak measurement for SSB spectra.
Accounts for the fact that SSB spectra have duplicated values in adjacent bins. Fixes #2282.
commit 764bce52 (2024-10-31): Fix spectrum peak measurement for SSB spectra. Fixes #2282.
closed 2024-10-31 by f4exb