f4exb/sdrangel #2294: Latest server version that works is 7.22.0

f4exb, 2024-10-28:
It appears that the latest version to work with the server flavour of SDRangel is 7.22.0. Something happened in 7.22.1 probably due to my changes for issue #2159 that caused an issue in the initialization sequence.

srcejon, 2024-10-28:
I would guess the server version needs a FSM to wait for the sync signals.

srcejon, 2024-10-31:
To recreate the problem, I'm running one of the example swagger scripts:

    ./rx_test.py -c -D RTLSDR

Part of the problem is in:

    void MainServer::changeSampleSource(int deviceSetIndex, int selectedDeviceIndex)

It does:

    deviceSet->m_deviceAPI->setSampleSource(source);
    deviceSet->m_deviceAPI->loadSamplingDeviceSettings(m_mainCore->m_settings.getWorkingPreset()); // load new API settings

Previously, setSampleSource would block, but this is no longer the case, so we have a race condition. 

We need to wait for the DSPDeviceSourceEngine::sampleSet signal before calling loadSamplingDeviceSettings.

We could do this with a FSM, as in MainWindow, but as MainServer.cpp is a bit simpler, we can get away with a single-shot lambda:

        // wait for sample source to be set, before loading settings
        auto connection = new QMetaObject::Connection();
        *connection = connect(deviceSet->m_deviceSourceEngine, &DSPDeviceSourceEngine::sampleSet, this, [=]() {
            deviceSet->m_deviceAPI->loadSamplingDeviceSettings(m_mainCore->m_settings.getWorkingPreset()); // load new API settings

            // Notify
            emit m_mainCore->deviceChanged(deviceSetIndex);
            
            QObject::disconnect(*connection);
            delete connection;        
        });       

Similar would be needed in changeSampleSink and changeSampleMIMO.

Qt6 has Qt::SingleShotConnection which would be nicer, as we don't need to create/delete the connection. but not in Qt5.  Might be better to code as an FSM for consistency with MainWindow however. But waiting for that signal fixes the problem for me.

f4exb, 2024-10-31:
Yes, this does fix the issue

f4exb, 2024-11-01:
Since it works I think this issue can be closed. Implementing FSM for the server is another subject. So far this solution is good enough.

closed 2024-11-01 by f4exb