f4exb/sdrangel #2296: chirp chat

DrSDR, 2024-10-29:
does anyone know if chirp chat can decode the meshtastic standard ?
if it can, what are the settings for long-fast channel?
here in the states the default channel is 906.875MHz  and use long-fast setting.
any help be greatly needed.
thanks again

f4exb, 2024-10-29:
You have the details of Meshtastic settings here: https://meshtastic.org/docs/overview/radio-settings/ For long fast it means 250k, SF=11 and FEC 4/5. SF=11 usually implies the " distance enhancement" so I think you should also have DE=2. With this information you should be able to parametrize the Chirp Chat demod. And it is LORA of course... 

However if it does work it will just dump the raw data so not as useful as the phone app... It may help analyzing the traffic though.

Hackrfan, 2024-10-31:
hello,
unfortunately the “Pre” setting is back to max 20 and not 32 as after the last update
greetings Georg

DrSDR, 2024-10-31:
thanks for input,  also what is the number of preambles for meshtastic long fast ?          and can we use the standard key of AQ==  somehow in chirpchat to decode the text ?

f4exb, 2024-11-02:
> hello,
> unfortunately the “Pre” setting is back to max 20 and not 32 as after the last update
> greetings Georg

You are probably using version 7.22.1.1 which is experimental and will soon be removed since it appears there is a better fix. You should stick to 7.22.2 in the meantime

Hackrfan, 2024-11-05:
Ok, what is the better solution? Is there a separate plugin for this?
vy 73 Georg

f4exb, 2024-11-06:
The purpose of the plugin or SDRangel is not to replace Meshtastic. It can be useful if you want to analyze what's going on on the air but for the Meshtastic functionality just use Meshtastic.

closed 2024-12-23 by github-actions[bot]