f4exb/sdrangel #2300: Apple Silicon M support

treaves, 2024-10-31:
I can compile and run sdrangle from the command line, but the package fails.  In looking through (auto closed) tickets, I saw someone used to create the Apple ARM packages, but they smashed their screen and is now unable to do so.  I'm willing & able to do so, but, the default cmake fails, so something else was being done.  If someone can fill me in on this, I don't mind helping out.

srcejon, 2024-10-31:
I'm trying to get the build working as a github action. First test build is here:

https://github.com/srcejon/sdrangel/actions/runs/11620019446/artifacts/2130848847

Please give it a try and let me know what happens. Be aware, it may not work at all. I can't test it at the moment.

treaves, 2024-11-01:
It seems to work for the basic usage.  And the GitHub action is the correct way forwards.  So thanks for that. Please let me know how I might help.

bazineta, 2024-11-01:
Tested on an M3, Sonoma 14.7.1, no issues so far.

closed 2024-11-01 by treaves

srcejon, 2024-11-01:
Thanks for trying it. 

There's a new build on the releases page that should include SDRplay, UHD and DATV support.