f4exb/sdrangel #2308: PlutoSDR plugin and FIR decimation

dm6tt, 2024-11-03:
Hi Edouard,
I'm playing around with the PlutoSDR plugin on 7.22.0 an have a question about the FIR decimation.
When enabling the FIR and setting the decimation to 4 I do not encounter a reduction of the ethernet bandwidth on that interface. It stays constant.

Referring to the RX Signal Path section in [https://wiki.analog.com/resources/tools-software/linux-drivers/iio-transceiver/ad9361](url) the FIR decimation factor can be querried on cli with:
`pluto:/sys/bus/iio/devices/iio:device0]# cat rx_path_rates`

Example:
Wit disabled FIR:
[image]

Leads to: 
`[pluto:/sys/bus/iio/devices/iio:device0]# cat rx_path_rates`
`BBPLL:953601026 ADC:29800032 R2:9933344 R1:4966672 RF:2483336 RXSAMP:2483336`

With enabled FIR:
[image]

Leads to:[pluto:/sys/bus/iio/devices/iio:device0]# cat in_voltage_sampling_frequency
2483336

`[pluto:/sys/bus/iio/devices/iio:device0]# cat rx_path_rates`
`BBPLL:1271468041 ADC:79466752 R2:39733376 R1:19866688 RF:9933344 RXSAMP:2483336`

In both cases the sampling frequency shows the same value:
`[pluto:/sys/bus/iio/devices/iio:device0]# cat in_voltage_sampling_frequency`
`2483336`

I would have expected that the Baseband sampling rate would be decreased with the FIR decimation.

Thanks
Marcus, DM6TT

f4exb, 2024-11-04:
That's the AD chip business not SDRangel's. In fact the baseband rate stays constant but the actual sampling frequency of the ADC is 4 times higher with a decimation of 4. It shows in the top left corner number below the start/stop button. This displays the actual ADC rate. See: https://github.com/f4exb/sdrangel/tree/master/plugins/samplesource/plutosdrinput#14-adc-sample-rate

f4exb, 2024-11-04:
This is actually explained in the link to the AD documentation you gave. In this table:
[image]
The HW decimation control only controls the decimation of the FIR filter so RF/ RXSAMP and effectively they are the same with a factor of 1: `RF:2483336 RXSAMP:2483336` and RF is 4 times RXSAMP when HW decimation is 4: `RF:9933344 RXSAMP:2483336` The rest of the chain is not controlled by the user.

f4exb, 2024-11-04:
Note that if the ethernet bandwidth is your concern you may use software decimation.

closed 2024-12-22 by github-actions[bot]